DanielOgorchock/linux #1: Rumble Support Not Working in fftest

e88z4, 2019-02-26:
Hi, 

I compiled your latest code with rumble support using nintendo switch pro controller. 

fftest shows the following capabilities

Device /dev/input/event16 opened
Features:
  * Absolute axes: X, Y, RX, RY, 
    [1B 00 00 00 00 00 00 00 ]
  * Relative axes: 
    [00 00 ]
  * Force feedback effects types: Periodic, Rumble, Gain, 
    Force feedback periodic effects: Square, Triangle, Sine, 
    [00 00 00 00 00 00 00 00 00 00 03 07 01 00 00 00 ]
  * Number of simultaneous effects: 16

Setting master gain to 75% ... OK
Uploading effect #0 (Periodic sinusoidal) ... OK (id 0)
Uploading effect #1 (Constant) ... Error: Invalid argument
Uploading effect #2 (Spring) ... Error: Invalid argument
Uploading effect #3 (Damper) ... Error: Invalid argument
Uploading effect #4 (Strong rumble, with heavy motor) ... OK (id 1)
Uploading effect #5 (Weak rumble, with light motor) ... OK (id 2)

When I tested #0, #4, #5, none of them provided any feedback. Do you know how I can test your code?

e88z4, 2019-02-26:
The following lines can be found in dmesg:

[Feb25 19:39]  (null): strong_mag=49152 weak_mag=49152
[  +0.055631]  (null): strong_mag=49152 weak_mag=49152
[  +0.056019]  (null): strong_mag=49152 weak_mag=49152
[  +0.055986]  (null): strong_mag=49152 weak_mag=49152
[  +0.056226]  (null): strong_mag=49152 weak_mag=49152
[  +0.055776]  (null): strong_mag=49152 weak_mag=49152
[  +0.055997]  (null): strong_mag=49152 weak_mag=49152
[  +0.056003]  (null): strong_mag=49152 weak_mag=49152
[  +0.055995]  (null): strong_mag=49152 weak_mag=49152
[  +0.055998]  (null): strong_mag=49152 weak_mag=49152
[  +0.056004]  (null): strong_mag=49152 weak_mag=49152
[  +0.056000]  (null): strong_mag=49152 weak_mag=49152
[  +0.056003]  (null): strong_mag=49152 weak_mag=49152
[  +0.055998]  (null): strong_mag=49152 weak_mag=49152
[  +0.056008]  (null): strong_mag=49152 weak_mag=49152
[  +0.055991]  (null): strong_mag=49152 weak_mag=49152
[  +0.056009]  (null): strong_mag=49152 weak_mag=49152
[  +0.056004]  (null): strong_mag=49152 weak_mag=49152
[  +0.055990]  (null): strong_mag=49152 weak_mag=49152
[  +6.286342]  (null): strong_mag=49152 weak_mag=49152
[  +1.025756]  (null): strong_mag=65535 weak_mag=49152
[  +3.807240]  (null): strong_mag=65535 weak_mag=65535
[  +1.280809]  (null): strong_mag=49152 weak_mag=65535
[  +3.840031]  (null): strong_mag=49152 weak_mag=49152
[  +1.760013]  (null): strong_mag=49152 weak_mag=49152
[  +0.056014]  (null): strong_mag=49152 weak_mag=49152
[  +0.055985]  (null): strong_mag=49152 weak_mag=49152
[  +0.056391]  (null): strong_mag=49152 weak_mag=49152
[  +0.055600]  (null): strong_mag=49152 weak_mag=49152
[  +0.056399]  (null): strong_mag=49152 weak_mag=49152
[  +0.055616]  (null): strong_mag=49152 weak_mag=49152
[  +0.056001]  (null): strong_mag=49152 weak_mag=49152
[  +0.055990]  (null): strong_mag=49152 weak_mag=49152
[  +0.056005]  (null): strong_mag=49152 weak_mag=49152
[  +0.056000]  (null): strong_mag=49152 weak_mag=49152
[  +0.055994]  (null): strong_mag=49152 weak_mag=49152
[  +0.055995]  (null): strong_mag=49152 weak_mag=49152
[  +0.056002]  (null): strong_mag=49152 weak_mag=49152
[  +0.056008]  (null): strong_mag=49152 weak_mag=49152
[  +0.055996]  (null): strong_mag=49152 weak_mag=49152
[  +0.056012]  (null): strong_mag=49152 weak_mag=49152
[  +0.055991]  (null): strong_mag=49152 weak_mag=49152
[  +0.024005]  (null): strong_mag=0 weak_mag=0

e88z4, 2019-02-26:
Nevermind about this issue. I looked at your code, it seems the functionality is not yet implemented. I will close this ticket.

closed 2019-02-26 by e88z4

DanielOgorchock, 2019-03-10:
Yeah sorry I'd been pushing unfinished work. I just got rumble working, so the latest patch should support it.

e88z4, 2019-03-11:
Hi, 

I tested your latest patch. Rumble is working. Thank you for your awesome work. I am waiting for your driver to support gyro.