DanielOgorchock/linux #2: More of a request....

GavinDarkglider, 2019-05-30:
In the version of hid-switchcon used in L4T-Switch(https://gitlab.com/switchroot/l4t-kernel-4.9) has the option to link joycons as 1 controller in the driver. Unfortunatly, this is needed if you want to use 2 joycons as one using kodi-game-libretro, and retroarch. Is there any chance we can get a patch for that feature for hid-joycon, which supports rumble, battery, and led changes? It would be nice to get all of these features in 1 driver, without having to have a complicated userspace application as a go between.

DanielOgorchock, 2019-05-30:
The folks on the kernel mailing list preferred userspace handling the joycon pairing. That said, if I get a chance to work on the driver this weekend I can add a patch that re-adds the kernel-side pairing support. Just know that that patch is unlikely to make it's way to mainline. I'm also hoping to get around to adding gyro support soon.

GavinDarkglider, 2019-05-30:
That is fine, for l4t, I dont think we care. lol. I havnt found a way to pair them in userspace as of yet, on my kodi gentoo build for my switch, and figured it would be easier to ask you to reapply the patch, then me trying to figure out what is going on with hid-switchcon, vs the latest version of hid-joycon. lol. Gyro support would be cool as well. I get if you dont post the pairing patch here, but if you could create the patch, I will get it to the L4T devs, so we can update the driver there back to hid-joycon, and keep the functionallity that is already there, and get back the functionallity that was added in hid-joycon.

DanielOgorchock, 2019-05-31:
I've looked at adding that functionality back tonight, and it seems like to would create quite a mess of things (there's so many more places now that special checks would be necessary with the rumble support and whatnot). I think the way to go is to have a userspace daemon which handles all the pairing using libevdev. When joy-cons are paired, it can create a virtual input device using libevdev's uinput support. I might take a stab at implementing something like that if I have the time. I think it will be much cleaner than having it in the driver, and it would also be useful for people using the mainline driver (assuming I finish it and actually get it mainlined sometime).

nicokoch, 2019-06-02:
I'm also currently working in such a user space driver. It is in its early stages and thus not public yet, but I can link it in this issue when it's ready.

DanielOgorchock, 2019-06-03:
I've written a userspace daemon called joycond which you can find here: https://github.com/DanielOgorchock/joycond

It will combine two joy-cons into a single uinput device. There's instructions on how to use it in its README. Be sure to use the latest version of the hid-joycon kernel driver from this repo if you want to try it out.

If this seems like something you'll want for the switch l4t project, I can look into creating an aur package or a .deb or something.

Note that I haven't gotten rumble working with the combined joy-cons yet, but I'll look into doing that soon.

DanielOgorchock, 2019-06-13:
@GavinDarkglider Does this seem like a suitable solution, or do you think support for it in the kernel is still needed?

GavinDarkglider, 2019-06-15:
This seems more than suitable to me. I can make it work, I dont use systemd in my build, but 1 script isnt much to contend with. lol. I am sure bylaws would also appreciate a deb file for the switchroot repo.

closed 2019-06-15 by GavinDarkglider