DanielOgorchock/linux #3: Backporting driver to l4t 4.9

GavinDarkglider, 2019-06-20:
I am getting these errors when trying to compile hid-joycon on l4t kernel 4.9:

drivers/hid/hid-joycon.c:223:1: error: initializer element is not constant
 joycon_rumble_amplitudes[ARRAY_SIZE(joycon_rumble_amplitudes) - 1].amp;
 ^~~~~~~~~~~~~~~~~~~~~~~~
drivers/hid/hid-joycon.c: In function ‘joycon_leds_create’:
drivers/hid/hid-joycon.c:1045:46: error: ‘LED_ON’ undeclared (first use in this function)
   led->brightness = ((i + 1) <= input_num) ? LED_ON : LED_OFF;
                                              ^~~~~~
drivers/hid/hid-joycon.c:1045:46: note: each undeclared identifier is reported only once for each function it appears in
make[2]: *** [scripts/Makefile.build:297: drivers/hid/hid-joycon.o] Error 1
 
I am not sure if there is something I am missing in the kernel config, but I am using the default L4T kernel config from switchroot, excepting the removal of hid-switchcon and the adding of hid-joycon. This happens on every version I have tested with led and rumble support. :(

GavinDarkglider, 2019-06-22:
I managed to get it to compile, with a few minor changes. First being that LED_ON was never defined in kernel 4.9(Patch here http://lkml.iu.edu/hypermail/linux/kernel/1701.0/02074.html)

The other feature with the rumble motor, I decided to hardcode the value from the array. Not the cleanest way to do it, but it seems to fix the error.

DanielOgorchock, 2019-06-22:
I'm glad you've gotten it compiling. Out of curiosity which gcc version are you using? I'm wondering why I'm not getting the "initializer element is not constant" error.

Please let me know if you manage to get it working on a 4.9 kernel. Billy was having issues getting calibration data responses on a 4.9 android kernel, and it would be good to know if the problem is android-specific.

Thanks.

GavinDarkglider, 2019-06-22:
Will let you know, I am using gcc-linaro-6.4.1-2018.05-x86_64_aarch64-linux-gnu to compile the kernel. I havnt managed to get a build that will boot, once I get it to compile. I think that has something to do with something else that gets pulled in as a dependency. I have no uart adapter, and no output on screen, and the logs are uselessly filled with debug data for bluetooth. Need more time to mess with it. to be certain though.

GavinDarkglider, 2019-07-14:
So, Figured I would update this, as I have finally managed to get a building version, that allows joycons to pair. 
First, LED's arent working(Not listed in /sys/class/leds, and when the controller is connected the lights stay the same as they are in pairing mode.

Second: Joycond is not recognising the driver, I am not sure if the needed code for that is in the wip code here, as I didnt apply that patch, I stopped at rumble support. The daemon runs, but is not linking the controllers together, and if it is, I cant tell, as there is no output from the app, and the leds dont change, and cant be configured. Let me know if you are interested in looking into it, and I will send you a copy of the kernel and boot chain I am using, and you can test it out on l4t-ubuntu.(Assuming you have a switch with RCM support, or a way to boot atmosphere, to reboot into hekate)

DanielOgorchock, 2019-07-14:
The WIP patch shouldn't be required for anything. The lack of LEDs is problematic. Is there any hid-joycon related dmesg output? I don't know how the driver could successfully probe without creating the led classdevs. It might be failing in the same place it was for Billy (timing out to retrieve the calibration response).

As long as the joycond udev rules are installed, joycond should output some debug messages every time it discovers a new device. I suspect the driver probe is failing.

I am interested in looking into it. I have a switch with RCM. Thanks!

GavinDarkglider, 2019-07-15:
Here is what I have, instructions are included. you will need l4t-ubuntu, and replace the kernel with this version. I have tested booting, and it boots, but I am having the same problem. Should also note, since I didnt in the readme, I am using the default tegra210-icosa.dtb, as it has not changed for this build, and for some reason, the new built version doesnt boot, with this version of coreboot.

GavinDarkglider, 2019-07-15:
[Hid-joycon.tar.gz](https://github.com/DanielOgorchock/linux/files/3393005/Hid-joycon.tar.gz)

also dont forget to apply switchroot update 1.3 first.

GavinDarkglider, 2019-07-23:
Just checking to see if there has been any progress with this issue.

DanielOgorchock, 2019-07-25:
Sorry for the delayed response. I haven't had a chance to take a look yet. I just downloaded the tarball and plan to put your build on my switch tonight.

DanielOgorchock, 2019-07-26:
Small update: I've been trying to get the l4t kernel building, but it seems like my toolchain is too new (so many warnings as errors). I'll try using an older toolchain this weekend.

bylaws, 2019-07-26:
Hey, android Dev here. 4.9 lacked driver autodetection and it was using the generic one instead of hid-joycon. Fixed that with patchset here https://gitlab.incom.co/CM-Shield/android_kernel_nvidia_linux-4.9_kernel_kernel-4.9/commits/lineage-16.0-icosa
However this still fails with android and no sent commands ever get a response.

bylaws, 2019-07-26:
If you want to debug the android issue it's probably easier to do on a phone, add in the kernel driver and modify the lineage bt stack till you fix it. But if you want to debug with switch android I can help you setup your environment so you can compile. Or if you want me to grab logs if anything specific I can.

GavinDarkglider, 2019-07-29:
I have this working with joycond in gentoo using bylaws's patchset. Should work for l4t ubuntu as well. I will leave this open, for the android bug that bylaws mentions above.

DanielOgorchock, 2019-07-30:
Great. Glad it's working and is narrowed down to android. @bylaws Thanks, I've been meaning to put lineage on my phone, so this would be a good excuse to do it.

DanielOgorchock, 2019-12-06:
Closing this because it's now understood to be an android BT stack issue which is fixed in Android 10.

closed 2019-12-06 by DanielOgorchock