DanielOgorchock/linux #5: Pro Controller disconnects with HZ=1000

Azenet, 2020-01-29:
Hi,

First, thanks for your work, it's great to be able to use the controller. I am however getting an issue: the controller usually fails to stay connected with my HZ=1000 kernel, while it works fine on HZ=300. The blue LED lights up for a split second, then turns off, and the controller works for about 2-3 seconds before turning off. The input device still exists and is stuck to the values the controller had before disconnecting.

<details>
<summary>Kernel logs</summary>

```
[ +11,657309] usb 4-1.2.1: new full-speed USB device number 34 using ehci-pci
[  +0,094944] usb 4-1.2.1: New USB device found, idVendor=057e, idProduct=2009, bcdDevice= 2.00
[  +0,000002] usb 4-1.2.1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  +0,000001] usb 4-1.2.1: Product: Pro Controller
[  +0,000001] usb 4-1.2.1: Manufacturer: Nintendo Co., Ltd.
[  +0,000001] usb 4-1.2.1: SerialNumber: 000000000001
[  +0,001442] nintendo 0003:057E:2009.0016: probe - start
[  +0,000169] nintendo 0003:057E:2009.0016: hidraw8: USB HID v81.11 Joystick [Nintendo Co., Ltd. Pro Controller] on usb-0000:00:1d.0-1.2.1/input0
[  +0,270882] nintendo 0003:057E:2009.0016: detected USB controller
[  +0,437542] nintendo 0003:057E:2009.0016: synchronous send/receive timed out
[  +0,000002] nintendo 0003:057E:2009.0016: retrying sync send after timeout
[  +0,110998] nintendo 0003:057E:2009.0016: synchronous send/receive timed out
[  +0,000002] nintendo 0003:057E:2009.0016: send usb command failed; ret=-110
[  +0,000001] nintendo 0003:057E:2009.0016: requesting cal data
[  +0,027453] nintendo 0003:057E:2009.0016: calibration:
              l_x_c=1989 l_x_max=3507 l_x_min=420
              l_y_c=1936 l_y_max=3493 l_y_min=311
              r_x_c=1905 r_x_max=3425 r_x_min=301
              r_y_c=2105 r_y_max=3637 r_y_min=475
[  +0,000002] nintendo 0003:057E:2009.0016: requesting IMU cal data
[  +0,023996] nintendo 0003:057E:2009.0016: IMU calibration:
              a_o[0]=46 a_o[1]=-160 a_o[2]=242
              a_s[0]=16384 a_s[1]=16384 a_s[2]=16384
              g_o[0]=-7 g_o[1]=-36 g_o[2]=-91
              g_s[0]=13371 g_s[1]=13371 g_s[2]=13371
[  +0,000001] nintendo 0003:057E:2009.0016: setting controller report mode
[  +0,032013] nintendo 0003:057E:2009.0016: enabling rumble
[  +0,023989] nintendo 0003:057E:2009.0016: enabling IMU
[  +0,055999] nintendo 0003:057E:2009.0016: controller MAC = 04:03:D6:44:38:EB
[  +0,000002] nintendo 0003:057E:2009.0016: setting player leds
[  +0,024085] nintendo 0003:057E:2009.0016: setting home led brightness
[  +0,024022] input: Nintendo Switch Pro Controller as /devices/pci0000:00/0000:00:1d.0/usb4/4-1/4-1.2/4-1.2.1/4-1.2.1:1.0/0003:057E:2009.0016/input/input59
[  +0,000091] input: Nintendo Switch Pro Controller IMU as /devices/pci0000:00/0000:00:1d.0/usb4/4-1/4-1.2/4-1.2.1/4-1.2.1:1.0/0003:057E:2009.0016/input/input60
[  +0,000047] nintendo 0003:057E:2009.0016: probe - success
(about 1 second later, the controller player LEDs turn off, then turn back on, the blue light flashes, then about 2 seconds later the player LEDs turn off and it stops responding)
[ +15,323796] usb 4-1.2.1: USB disconnect, device number 34 (disconnected the cable)
[  +0,000051] nintendo 0003:057E:2009.0016: remove
[  +0,044432] nintendo 0003:057E:2009.0016: setting home led brightness
[  +0,000018] nintendo 0003:057E:2009.0016: setting player leds
[  +0,000015] nintendo 0003:057E:2009.0016: setting player leds
[  +0,000014] nintendo 0003:057E:2009.0016: setting player leds
[  +0,000014] nintendo 0003:057E:2009.0016: setting player leds
```

</details>

I've had no success trying to tune the timeouts to make it work, unfortunately. I've tried different values and turning up the number of retries to no change.

Thank you!

DanielOgorchock, 2020-01-29:
Do you happen to have Steam running during the failure case? I've seen similar behavior due to Steam using hidraw to directly implement its own userspace driver, and it competes with the kernel driver.

Azenet, 2020-01-29:
You're right, it is Steam causing it, closing it works. Thanks!

closed 2020-01-29 by Azenet

DanielOgorchock, 2020-01-29:
I documented a workaround on the archwiki here: https://wiki.archlinux.org/index.php/Gamepad#Using_hid-nintendo_with_Steam_Games

Azenet, 2020-01-29:
Thanks again, that works for me.