DanielOgorchock/linux #6: hid-nintendo as DKMS module?

dragon99919, 2020-03-16:
I've tried to use the joycond made by you in my Manjaro setup, but after failing, I've noticed that the hid-nintendo module is still not yet in the kernel officially. Since I'm sort of noob with any of the building stuff, are there any chances this gets ported temporarly as DKMS module? Or maybe could you provide instructions on how to handle something like that? Great works btw, best regards!

DanielOgorchock, 2020-03-17:
I have considered adding an out-of-tree module build option in the past. I just submitted a new patchset to the mailing list. I suspect it's pretty close to being merged in. I've removed the IMU patch for now, since that's where most of the needed work is.
If the feedback on my latest patchset looks like it'll be awhile still, I may mess around with an out-of-tree option to put on the aur in the meantime.

nicman23, 2020-03-17:
https://github.com/nicman23/dkms-hid-nintendo

i did a junk dkms module that works - 0 idea how to make it default to the hid-nintendo

currently needs something like 

echo 0003:057E:2009.000A > /sys/class/input/js0/device/device/driver/unbind
echo 0003:057E:2009.000A > /sys/bus/hid/drivers/nintendo/bind

DanielOgorchock, 2020-03-17:
Cool. Thanks.

mattrei, 2020-03-25:
do you have any readme how to install the dkms module? 
Yes an AUR package would be fantastic as well!

nicman23, 2020-03-25:
clone the repo; cd into it
```
sudo dkms add .; sudo dkms build nintendo -v 1.0
```

it is generic so i did not bother with it

mattrei, 2020-03-26:
cheers!

the gyroscope of the joycons are not working yet with the driver, right?

nicman23, 2020-03-26:
> the gyroscope of the joycons are not working yet with the driver, right?

it does but it is a different device

also just pushed hid-nintendo-dkms to the aur

mattrei, 2020-03-26:
@nicman23  thanks for you work! 

i have connected succesfully the joycons via bluetooth and _evtest_ lists them as devices. however the gyro does not raise any events. also I have no extra device listed.

may you give a short tutorial how to get the gyro  running? thx

nicman23, 2020-03-26:
no idea - i am not the original creator and have only messed with the pro controller

if you get it running please tell me how

DanielOgorchock, 2020-03-26:
@mattrei Are you running evtest as root? I think the IMU evdevs don't get picked up by uinput for changing permissions.

Note that I haven't done much testing yet with the gyro support, and I removed it from the latest submission to the mailing list. I do have that in-progress patch still up in this repo, which in theory works, but there's some issues I need to fix with it still.

DanielOgorchock, 2020-03-26:
@nicman23 Thanks for the aur package!

nicman23, 2020-03-26:
>Thanks for the aur package!

no prob :)

>Note that I haven't done much testing yet with the gyro support, and I removed it from the latest submission to the mailing list. I do have that in-progress patch still up in this repo, which in theory works, but there's some issues I need to fix with it still.

fyi the dkms modules has it

mattrei, 2020-03-26:
@DanielOgorchock  oh yes the Joy-Con IMU is popping up on _evtest_ with root privileges. is there a way to have it also for non-root users?

my main wish for getting joycons gyro to work is playing Super Mario Galaxy on the dolphin emulator on linux :100:

dogtopus, 2020-03-27:
@mattrei You can set a udev rule (recommended) or put yourself in the input group (not recommended for daily use since it also exposes keyboard, etc to all user apps)

DanielOgorchock, 2020-03-27:
I should probably add that to the joycond udev rules

closed 2020-04-06 by DanielOgorchock

mpawlowski7, 2020-05-22:
Hi, If anyone is still interested I did a dkms module for this driver a while ago adding udev rules and sdl2 mapping:
https://github.com/mpawlowski7/NXControllers
I also improved player led counting so it's now working as on the switch.