DanielOgorchock/linux #8: Calibration issues with Right Joycon

joaorb64, 2020-04-17:
I'm opening this issue more as a way to research and discuss a solution, or find something I'm missing. While developing joycond-cemuhook, I'm noticing that specifically the Right Joycon IMU registers greater values for the axis -ABS_RZ (using it as a wiimote, it's the motion as to move the pointer to the left).

An example of this issue would be that using a Pro Controller as a pointer won't lose center too quickly, but with a Right Joycon the small readings from me just holding it still will bring the pointer mostly towards the left and a bit down.

I've tried using some sort of deadzone but even then as it reads rotation values these axes' positive and negative readings mismatch by a lot.

Could it be a calibration issue? Or am I missing something?

DanielOgorchock, 2020-04-17:
I'm not sure if this is the issue you're seeing, but have you taken the reversed axes orientations of the right joy-con into account (as seen here: https://github.com/dekuNukem/Nintendo_Switch_Reverse_Engineering/blob/master/imu_sensor_notes.md)?

The right joy-con's axes have a different orientation compared to the left joy-con's and pro controller's. One of the things I have on the to-do list for hid-nintendo is to do the correction in the driver instead of making userspace handle it.

joaorb64, 2020-04-17:
Yes, I made my code so that it selects the correct axis and the correct signal for them.
The problem itself is that the Right Joycon, even if on a flat surface, reads a lot of noise mainly coming as -ABS_RZ and, even if I use a deadzone, it reads greater values for -ABS_RZ than for +ABS_RZ and loses its reference point anyways.
I'm reading events from evdev.

DanielOgorchock, 2020-06-01:
I don't know if you'd already found a workaround to this issue, but the latest patchset has support for reading IMU user calibrations. I wonder if calibrating on the switch and then retrying with the driver would alter behavior.

joaorb64, 2020-06-01:
Thanks for the heads up. I'll look into it probably next week, but it does sound like this could be a fix for my issue.

joaorb64, 2020-06-19:
I tested it and I was having the same results. Then, calibrated the joycon on the Switch and tested again, and it got better! The cursor on Dolphin was going down and left, now it was going only to the right but at a slower rate. Then, testing again using Splatoon 2 on the Switch, I noticed it can have this same problem, but I guess the software deals with it better. I'll try messing more with deadzones for gyro data. Thanks!

closed 2020-06-19 by joaorb64