DanielOgorchock/linux #11: nintendo-hid IMU device screws up joystick order in udev

buddwm, 2020-06-02:
I've been using my Nintendo Switch Pro controller via bluetooth in Ubuntu 20.04.  When using this driver, I noticed that it creates a joystick device for the controller, and one other separate device for the IMU.  This has led to numerous issues in programs such as Retroarch and PCSX2 in which the 2nd player controller is always auto selected as the IMU device instead of a 2nd controller paired via bluetooth.  Programs like PCSX2 will obey the order of joysticks as defined in /etc/input.  This is what joysticks get listed when using this driver (as an example):

crw-rw----+  1 root input 13,   0 Jun  1 23:24 js0
crw-rw----   1 root input 13,   1 Jun  1 23:24 js1
crw-rw----+  1 root input 13,   2 Jun  1 23:31 js2
...
lrwxrwxrwx   1 root root        3 Jun  1 23:24 nintendo-controller-1 -> js0
lrwxrwxrwx   1 root root        3 Jun  1 23:31 xbox-controller-1 -> js2

Notice that my xbox controller becomes js2 instead of js1.  This is because js1 is the IMU device.

The desirable behavior would be to not list the IMU as a separate joystick device.  I've had to modify the driver and remove any reference to the IMU stuff as a workaround for the time being.  I understand that the IMU is supposed to add the gyroscope capabilities of the controller, so would be nice to have working, but having the joysticks out of order is a show-stopper for the automation around a lot of my games, and most programs will try to use the joystick device as an actual controller.

DanielOgorchock, 2020-06-06:
Do you happen to have a dualshock controller with an IMU? I based this implementation on the hid-sony driver, and I'm curious if that driver has the same symptoms you're seeing.

DanielOgorchock, 2020-06-06:
After digging into it further, I figured it out. The IMU input device needs to be blacklisted from the joydev input handler. Please see https://github.com/DanielOgorchock/linux/commit/8b51e809be855ff23237dee866f5629f41bf916c#diff-34c8d0c14a101d038ef2d2e7a1957559

closed 2020-06-06 by DanielOgorchock

buddwm, 2020-06-06:
Thanks for the quick response!  I was able to compile the new joydev module and am getting the intended results.