DanielOgorchock/linux #12: Pro Controller not using driver

aanarchy, 2020-06-26:
I'm on Arch linux with the latest updates.

I've tried building the dkms module manually, using the AUR, nothing works.

The hid_nintendo doesn't pick up the controller. It defaults to the hid-generic driver, and if I try to unbind it from the hid-generic driver it goes through. 

Rebinding to the hid_nintendo driver though gives me this error:
`-bash: echo: write error: No such device`

nicman23, 2020-06-26:
is it an official controller?

aanarchy, 2020-06-26:
Wow, thanks for the quick response.
Yes, I'm using an official controller

nicman23, 2020-06-26:
hmm i know this is stupid but have you rebooted / run depmod?

aanarchy, 2020-06-26:
Yeah, multiple times. 110% sure that is not the issue. Besides, manually rebinding the controller doesn't work anyways.

aanarchy, 2020-06-27:
I checked modinfo and the modalias of my pro controller as well, and everything checks out. The controller should be correctly aliased to the hid_nintendo driver according to the information I found from `/sys/bus/hid/devices` and `modinfo hid_nintendo` but that is not the case.

aanarchy, 2020-06-27:
I'm going to try to rollback to the `linux-lts` kernel from the official arch repos, will report back later.

aanarchy, 2020-06-27:
Sorry for wasting your time, I forgot that I had set his to load with `ignore_special_drivers=1` to make DS4 compatible with RPCS3, my bad :D

closed 2020-06-27 by aanarchy