DanielOgorchock/linux #13: Resolve conflicts with Steam

nicman23, 2020-08-01:
can we mask hidraw (talking out of my ass - i have 0 idea about the subsystem) so that steam does not mess with the controllers without a messy jail?

clarfon, 2020-08-01:
This is a bug on Steam's end -- bother them at ValveSoftware/steam-for-linux#6651 to add proper detection. The short answer is no.

DanielOgorchock, 2020-08-02:
I have tried disabling hidraw in hid-nintendo in the past. Steam doesn't have any hidraw to mess with, but it doesn't seem to even bother trying to read the controller inputs with evdev. Someone from steam recently started a conversation on the mailing list about the overall situation here if you haven't seen it: https://www.spinics.net/lists/linux-input/msg68303.html

nicman23, 2020-08-02:
hmm maybe then we can do something in joycond? the `firejail --noprofile --blacklist=/sys/class/hidraw/ steam` workaround means that disallowing permissions should work

nicman23, 2020-08-02:
https://github.com/DanielOgorchock/joycond/pull/27

closed 2020-08-02 by nicman23