DanielOgorchock/linux #14: Support for the official NES and SNES controllers

photodiode, 2020-08-20:
Hi,
Would it be possible to add in support for the official NES and SNES controllers?

@nadiaholmquist added them into an older version of this driver which worked fine, so I'm guessing any needed information can be found in there. (Links to the code: [hid-ids.h](https://github.com/nadiaholmquist/linux/blob/21a824e0ac94b0e8453027ea22579e2d2f109ae3/drivers/hid/hid-ids.h), [hid-nintendo.c](https://github.com/nadiaholmquist/linux/blob/21a824e0ac94b0e8453027ea22579e2d2f109ae3/drivers/hid/hid-nintendo.c))

Thanks for all the work put into this driver,
it's immensely appreciated!

nadiaholmquist, 2020-08-20:
Once the driver makes it into mainline, I'm going to try and submit my patches to add support for those controllers too.

photodiode, 2020-08-20:
Thank you so much

v1993, 2021-02-01:
Relevant PR for SNES controller on DKMS version of this module: https://github.com/nicman23/dkms-hid-nintendo/pull/27

emilyst, 2022-03-28:
> Once the driver makes it into mainline, I'm going to try and submit my patches to add support for those controllers too.

It looks like the `hid-nintendo` driver [hit mainline in 5.16](https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/commit/?id=5af06603c4090617be216a9185193a7be3ca60af) (congrats, Daniel!).

I'm aware of @nadiaholmquist's existing work to add support for the [NES joycons](https://github.com/nadiaholmquist/linux/commit/21a824e0ac94b0e8453027ea22579e2d2f109ae3) and the [SNES controller](https://github.com/nadiaholmquist/linux/commit/bdf4626b45eb631b61a55817dcf789f81f9bddb0) against an older version of `hid-nintendo`. These changes work great when I test them locally.

I'd love to help move NES and SNES support ahead, if that's possible, against the latest version of `hid-nintendo`. What would that entail? Applying @nadiaholmquist's patches to `hid-nintendo` and opening a pull request for each here? Or do I need to interact with the Linux Input Mailing List?

In addition, I have the [Nintendo Switch Online Sega Genesis control pad](https://www.nintendo.com/store/products/sega-genesis-control-pad/). It has product ID `0x201e`, so it unfortunately gets picked up by the `hid-generic` driver and does not work at all (e.g., `jstest` reports no events). I'd be happy to attempt to write a patch for it as well. And if I ever manage to get ahold of the [N64 controller](https://www.nintendo.com/store/products/nintendo-64-controller/), I could try it as well.

Thanks to you both for the work you've done already. Let me know if or how I can help.

emilyst, 2022-03-28:
> I'd love to help move NES and SNES support ahead, if that's possible, against the latest version of `hid-nintendo`. What would that entail? Applying @nadiaholmquist's patches to `hid-nintendo` and [...]

FYI: I have a draft of this work done, available to view in this pull request: https://github.com/DanielOgorchock/linux/pull/35. I've marked it as a draft to indicate that it's not fully done, but it's ready for input if desired. Hopefully this will speed up getting this support incorporated into kernel 5.17 or 5.18.

If I don't receive a response, I'll try emailing you, @DanielOgorchock, or going to the Linux Input list directly.

emilyst, 2022-04-02:
I ended up closing that pull request since I got no response, but anyone who likes can install my work from here: https://github.com/emilyst/hid-nintendo/tree/v1.0.

nfp0, 2024-01-27:
Has there been any progress in getting the rest of the controllers mainlined?

SuperSamus, 2024-01-27:
https://github.com/torvalds/linux/commit/94f18bb19945915fcdfd1903841020ef1b6af44a
Will be merged for Linux 6.8.

nfp0, 2024-01-27:
Wow, I did not expect that. Great news!
Thank you for everyone involved!