DanielOgorchock/linux #15: Driver not loading for PowerA Enhanced Wireless Switch Pro Controller

KaryotypeB, 2020-10-15:
I'm not sure of what exactly the problem is, since the controller works fantastically on Steam, 
but logs seem to indicate the controller doesn't report its vendor or product id correctly in some way so the module isn't assigned to the controller?
I'm not familiar with the inner workings of the kernel so don't take my guess too seriously.
I've tried force loading the module with Steam closed before connecting the controller but no dice.
It may be a problem with my setup, I'm not too sure. If it's relevant, the SKU for the controller is `1513552-01`

Relevant journal log (force loaded the module before just in case):
```
Oct 15 14:38:26 olympus kernel: hid_nintendo: loading out-of-tree module taints kernel.
Oct 15 14:38:26 olympus kernel: hid_nintendo: module verification failed: signature and/or required key missing - tainting kernel
Oct 15 14:38:26 olympus sudo[11670]: pam_unix(sudo:session): session closed for user root
Oct 15 14:38:26 olympus audit[11670]: USER_END pid=11670 uid=0 auid=1000 ses=2 msg='op=PAM:session_close grantors=pam_limits,pam_unix,pam_permit acct="root" exe="/usr/bin/sudo" hostname=? addr=? terminal=/dev/pts/2 res=success'
Oct 15 14:38:26 olympus audit[11670]: CRED_DISP pid=11670 uid=0 auid=1000 ses=2 msg='op=PAM:setcred grantors=pam_faillock,pam_permit,pam_faillock acct="root" exe="/usr/bin/sudo" hostname=? addr=? terminal=/dev/pts/2 res=success'
Oct 15 14:38:26 olympus kernel: audit: type=1106 audit(1602794306.566:332): pid=11670 uid=0 auid=1000 ses=2 msg='op=PAM:session_close grantors=pam_limits,pam_unix,pam_permit acct="root" exe="/usr/bin/sudo" hostname=? addr=? terminal=/dev/pts/2 res=success'
Oct 15 14:38:26 olympus kernel: audit: type=1104 audit(1602794306.566:333): pid=11670 uid=0 auid=1000 ses=2 msg='op=PAM:setcred grantors=pam_faillock,pam_permit,pam_faillock acct="root" exe="/usr/bin/sudo" hostname=? addr=? terminal=/dev/pts/2 res=success'
Oct 15 14:38:41 olympus kernel: hid-generic 0005:0000:0000.000B: unknown main item tag 0x0
Oct 15 14:38:41 olympus kernel: input: Lic Pro Controller as /devices/pci0000:00/0000:00:01.3/0000:01:00.0/usb1/1-10/1-10:1.0/bluetooth/hci0/hci0:256/0005:0000:0000.000B/input/input46
Oct 15 14:38:41 olympus kernel: hid-generic 0005:0000:0000.000B: input,hidraw6: BLUETOOTH HID v0.01 Gamepad [Lic Pro Controller] on 1c:4d:70:05:73:72
```

Relevant Steam log entry:
```
Local Device Found
  type: 057e 2009
  path: /dev/hidraw6
  serial_number: 30:31:7d:12:1a:34 - 0
  Manufacturer:
  Product:      Lic Pro Controller
  Release:      0
  Interface:    -1

CApplicationManagerPopulateThread took 582 milliseconds to initialize (will have waited on CAppInfoCacheReadFromDiskThread)
Installing breakpad exception handler for appid(steam)/version(1602115886)
Proceed to auto login

** (steam:7543): WARNING **: 14:20:45.262: Could not initialize NMClient /org/freedesktop/NetworkManager: The name org.freedesktop.NetworkManager was not provided by any .service files
Switch: Bluetooth
!! Steam controller device opened for index 0.
Steam Controller reserving XInput slot 0
Installing breakpad exception handler for appid(steam)/version(1602115886)
Installing breakpad exception handler for appid(steam)/version(1602115886)
Controller 0 connected, configuring it now...
Installing breakpad exception handler for appid(steam)/version(1602115886)
Installing breakpad exception handler for appid(steam)/version(1602115886)
Switch Controller calibration:
  Sensor 0: bias -1, sensitivity 4.0000  Sensor 1: bias -1, sensitivity 4.0000  Sensor 2: bias 4097, sensitivity 4.0000  Sensor 3: bias 1, sensitivity 2.0000  Sensor 4: bias -10, sensitivity 2.0000  Sensor 5: bias 0, sensitivity 2.0000
!! Controller 0 attributes:
  Type: 38
  ProductID: 8201
  Serial: NLP30317d121a34
  Capabilities: 00856bff
  Firmware Version: 0
  Firmware Build Time: 2147483647 (Tue, 19 Jan 2038 03:14:07 GMT)
  Bootloader Build Time: 2147483647 (Tue, 19 Jan 2038 03:14:07 GMT)
```

Gibbz, 2020-11-13:
I have one of these controllers im trying to get working on the new 5.10 kernel. I can sync the bluetooth, but the controller is unusable and not visible to steam.
Do this require the use of joycond to work? Or should I simply be able to sync, and steam should pickup the device?
How can I help provide debug information for this?

Brando753, 2021-02-13:
Okay I was crazy trying to figure out why I could not get my controller to work. I can confirm this controller is also giving me problems on my setup with Debian Sid.

cipitaua, 2021-04-30:
I'm on ubuntu-devel, the nintendo-pro (original) controller connects via bluetooth correctly with the hid-generic:

```
[ 1375.472523] hid-generic 0005:0F0D:00F6.0003: unknown main item tag 0x0
[ 1375.472751] input: Lic Pro Controller as /devices/virtual/misc/uhid/0005:0F0D:00F6.0003/input/input23
[ 1375.474035] hid-generic 0005:0F0D:00F6.0003: input,hidraw0: BLUETOOTH HID v0.01 Gamepad [Lic Pro Controller] on d8:f2:ca:b8:f6:95
```
and it works fine, although the back leds keep blinking nonstop. I have installed dkms-hid-nintendo on kernel 3.11, and
after restarting the system I get the
`hid_nintendo: loading out-of-tree module taints kernel.`
message in dmesg.

However, the controller seems to be grabbed by hid-generic and not by hid-nintendo.

joaorb64, 2021-05-01:
@cipitaua the default hid-generic behavior doesn't have access to rumble or motion sensors, and the analog sticks won't ever register 100% right or up if I'm not mistaken

cipitaua, 2021-05-02:
indeed, so what can I do to use hid-nintendo instead of hid-generic?

cipitaua, 2021-05-03:
this are the `hid-*` modules loaded:

```
~$ lsmod | grep hid
hid_generic            16384  0
hid_nintendo           36864  0
ff_memless             20480  1 hid_nintendo
usbhid                 61440  0
uhid                   20480  1
i2c_hid                32768  0
hid                   135168  5 hid_nintendo,i2c_hid,usbhid,hid_generic,uhid
mac_hid                16384  0
```

hid-nintendo is present, but somehow it does not grab the ProController during connection

grinness, 2021-12-26:
Hi,

recently purchased one of the PowerA Nintendo Pro Controller -- my understanding been that it is supported by the dkms and upcoming hid-nintendo in 5.16

However, I am having some issues with it (some pointed out above)
my dmesg:

[ 593.532244] usb 3-2: new full-speed USB device number 3 using xhci_hcd
[  593.685175] usb 3-2: New USB device found, idVendor=20d6, idProduct=0002, bcdDevice= 1.00
[  593.685178] usb 3-2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  593.685179] usb 3-2: Product: NSW Enhanced Wireless Controller Rechargeable
[  593.685179] usb 3-2: Manufacturer: BDA
[  593.685180] usb 3-2: SerialNumber: LBI-969_v1.93
[  593.714265] input: BDA NSW Enhanced Wireless Controller Rechargeable as /devices/pci0000:00/0000:00:01.2/0000:02:00.0/0000:03:08.0/0000:07:00.3/usb3/3-2/3-2:1.0/0003:20D6:0002.0009/input/input28
[  593.714320] hid-generic 0003:20D6:0002.0009: input,hidraw7: USB HID v1.11 Gamepad [BDA NSW Enhanced Wireless Controller Rechargeable] on usb-0000:07:00.3-2/input0

If controller plugged in via USB-C, it is exposed in /dev/input/js* but it does not work (tested with jstest-gtk)

If controller connected in via BLUETOOTH, it is exposed in /dev/input/js* and it works (tested with jstest-gtk) 
However:
- no xinput interface
- analog sticks register up to ~85% 
- Steam is able to 'grab' the controller and works, but if running steam no other app can access the joy-pad

The controller is  not handled by the  dkms hid-nintendo
Below lsmod when connected via USB-C:
hid_nintendo           45056  0
hidp                   36864  0
bluetooth             778240  44 btrtl,hidp,btintel,btbcm,bnep,btusb,rfcomm
ff_memless             20480  2 hid_nintendo,xpad
hid_apple              20480  0
usbhid                 81920  0
mac_hid                16384  0

Below when connected via bluetooth:
lsmod | grep hid
hid_nintendo           45056  0
hidp                   36864  1
bluetooth             778240  38 btrtl,hidp,btintel,btbcm,bnep,btusb,rfcomm
ff_memless             20480  2 hid_nintendo,xpad
hid_apple              20480  0
usbhid                 81920  0
mac_hid                16384  0

Tried to build my own udev rules, no luck.
Any suggestion would be appreciated

Thanks