DanielOgorchock/linux #18: hid-nintendo not present in Linux 5.10?

ZLima12, 2020-12-16:
I've been hearing all over the place that hid-nintendo was merged in for Linux 5.10. I'm about to update, but I can't seem to see the driver in the tree. What gives?

nicman23, 2020-12-16:
let 5.10 move from testing to core in arch linux and i ll tell you :P

dogtopus, 2020-12-16:
It's not there yet. Maybe it got delayed again or it will be merged before 5.11 comes out and got backported to some revision of 5.10.

ZLima12, 2020-12-16:
Oh well. Looking forward to 5.11 then.

closed 2020-12-16 by ZLima12

gardotd426, 2021-02-16:
Not in 5.11 (or seemingly 5.12), either. It's still not even in `linux-next`. And no commits since January. Seems like the plans to merge are at least on hold

Dunexus, 2021-02-21:
Has this been reviewed or rejected by someone able to validate the merge ?

I don't know the process.

gardotd426, 2021-02-21:
Well, it's out for 5.12 at this point, but not 5.13. Basically anything like this that doesn't make it in before the merge window which ends right before the first release candidate comes out, doesn't make it at all. 

As far as this goes, it was in the `for-5.10` branch of the `hid` linux repo (which is where input-related kernel stuff happens) queued up for 5.10, but it never actually got merged. It's unknown right now (I guess except by maybe one or two people) why it didn't get merged, as there wasn't anything on the mailing list about it. I know Daniel (developer) had said he needed to do a bit more work, but this was before it was even put in the 5.10 repo, since then there's not been anything on the mailing list. 

Basically, either Daniel told the maintainer to hold off, or the maintainer decided it wasn't ready, or the maintainer forgot to include it, and until one of them decides to let us know (or merges it), that's all we'll know.

iMonZ, 2021-08-23:
Anything new here?

DanielOgorchock, 2021-09-02:
Hopefully in time for 5.15 or 5.16. There were delays while working out proper naming for the leds

iMonZ, 2021-09-02:
> Hopefully in time for 5.15 or 5.16. There were delays while working out proper naming for the leds

How far are you in the project? I think the merge window has already started.
I just hope this will be merged until the next Linux LTS.
I think It's either 5.15 or 5.16

gardotd426, 2021-09-02:
>How far are you in the project? I think the merge window has already started.

5.15's merge window is open right now, 5.16 hasn't opened yet.

Kleysley, 2021-10-31:
They say it will merge in Linux 5.16

iMonZ, 2021-11-04:
> They say it will merge in Linux 5.16

For the LTS it’s too late :(
Who said it will be merged in 5.16?

ZLima12, 2021-11-04:
> For the LTS it’s too late :(

Honestly most systems that one would play games on should just use the stable kernel, not LTS.

iMonZ, 2021-11-04:
> > For the LTS it’s too late :(
> 
> Honestly most systems that one would play games on should just use the stable kernel, not LTS.

Totally correct!

Dunexus, 2021-11-05:
> Who said it will be merged in 5.16?

'for-5.16/nintendo' with all the hid-nintendo work has been merged into 'for-next', see this [merge commit](https://git.kernel.org/pub/scm/linux/kernel/git/hid/hid.git/commit/?h=for-next&id=900acbbe2552b17c80718d7247fb59b3c104a4ba)