DanielOgorchock/linux #19: HID input reports are scrambled, causes problems in some wine applications

LunarLambda, 2021-01-18:
After some discussion [in this wine bug report](https://bugs.winehq.org/show_bug.cgi?id=50510) I was pointed toward hid-recorder, and indeed, hid-recorder (which reads /dev/hidraw3, the hid device assigned to my official switch pro controller), spews out a bunch of incoherent input data that seems to match the weird nonsensical data I got in some wine apps, particularly games using the windows HID APIs.

Since this tells me this isn't a wine-specific issue, I was hoping to get some guidance here. I'm using the hid-nintendo-dkms driver version 3.0.

the js/event devices strangely enough work. I'm wondering if this is because the raw HID data includes data from the IMU, which higher level systems like ev/js filter out? But that's just blind guessing, really.

Attached is a log from hid-recorder (which is part of libevdev/hid-tools)
[abc.txt](https://github.com/DanielOgorchock/linux/files/5831849/abc.txt)

nicman23, 2021-01-20:
@DanielOgorchock this probably interest you?

LunarLambda, 2021-02-04:
relevant kernel messages from /proc/kmsg:

```
<6>[81257.083856] usb 1-1: new full-speed USB device number 6 using xhci_hcd
<6>[81257.235710] usb 1-1: New USB device found, idVendor=057e, idProduct=2009, bcdDevice= 2.10
<6>[81257.235716] usb 1-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
<6>[81257.235751] usb 1-1: Product: Pro Controller
<6>[81257.235755] usb 1-1: Manufacturer: Nintendo Co., Ltd.
<6>[81257.235757] usb 1-1: SerialNumber: 000000000001
<6>[81257.244212] nintendo 0003:057E:2009.0006: hidraw3: USB HID v81.11 Joystick [Nintendo Co., Ltd. Pro Controller] on usb-0000:00:14.0-1/input0
<6>[81258.116675] nintendo 0003:057E:2009.0006: using factory cal for left stick
<6>[81258.154101] nintendo 0003:057E:2009.0006: using factory cal for right stick
<6>[81258.220679] nintendo 0003:057E:2009.0006: using factory cal for IMU
<6>[81258.357785] nintendo 0003:057E:2009.0006: controller MAC = E0:F6:B5:66:6F:89
<6>[81258.412889] input: Nintendo Switch Pro Controller as /devices/pci0000:00/0000:00:14.0/usb1/1-1/1-1:1.0/0003:057E:2009.0006/input/input25
<6>[81258.413214] input: Nintendo Switch Pro Controller IMU as /devices/pci0000:00/0000:00:14.0/usb1/1-1/1-1:1.0/0003:057E:2009.0006/input/input26
```

I noticed that `/sys/devices/pci0000:00/0000:00:14.0/usb1/1-1/1-1:1.0/0003:057E:2009.0006` has 2 devices under input/ but only one under hidraw/

Also I can't find anything corresponding to `usb-0000:00:14.0-1/input0` as mentioned in the hidraw3 line.

LunarLambda, 2021-02-26:
I've found out what the actual issue is.

When connected via USB, the controller sends an incorrect HID descriptor, causing anything that tries to parse the input data with it to report total gibberish.

See this thread:
https://gbatemp.net/threads/reverse-engineering-the-switch-pro-controller-wired-mode.475226/

The person in that thread made a custom parser that works with USB:
https://github.com/ToadKing/switch-pro-x/blob/master/switch-pro-x/ProControllerDevice.cpp#L80
https://github.com/ToadKing/switch-pro-x/blob/master/switch-pro-x/ProControllerDevice.cpp#L194
https://github.com/ToadKing/switch-pro-x/blob/master/switch-pro-x/ProControllerDevice.cpp#L254

I'm not super familiar with how the HID stack works, could this be fixed?

LunarLambda, 2021-07-31:
I've been looking at the code and it appears that the code pertaining to the /dev/input/js interface (which works perfectly even in USB mode) working is down to the fact that rather than working off the bogus HID descriptor reported by the controller, it simply hardcodes the input structure and the offsets used for hid_field_extract.

The realization I made from this is that, despite the name 'hid-nintendo', the driver has absolutely no business with the hidraw or hiddev systems (and therefore cannot 'fix' applications that use those systems), and purely provides a (working) input/evdev implementation.

With that knowledge, I'm closing this issue.

closed 2021-07-31 by LunarLambda