DanielOgorchock/linux #20: Gyro resolution appears to be off

v1993, 2021-02-01:
I've tested a pair of joycons and a pro controller. When using their gyro information, all of them end up reporting a single revolution as a single revolution plus about 30-45 degrees. Since offset is almost exactly the same for every tested device, I have to guess that default gyro resolution is incorrect.

v1993, 2021-02-05:
After some testing, it appears that gyro values are off by about 0.858 times.

joaorb64, 2021-02-05:
A few days ago I tried playing Wii Sports Resort using joycond-cemuhook and it didn't work out as well as I expected it to be. I didn't have the rumble fix applied and I'm pretty sure it made it worse, but I'm inclined to agree that gyro data could be a bit off. To be honest I'm coming from a context where multiple layers of software are involved so it could be any of them though.

v1993, 2021-02-05:
@joaorb64 You don't actually need `joycond-cemuhook` (or `evdevhook`, for the record :wink:) to use this driver with dolphin, you can bind motion directly. Also, you can use those view monitors in dolphin (one for acceleration, one for gyro) to look at results. I've found them to be the best source of movement calibration when working on various motion-related stuff.

hoopahmadness, 2021-02-26:
I'm sorry to ask this here but I just read @v1993  saying that the correct way to use joycon with Dolphin is to bind them directly. I've been having a ton of trouble with that, the devices that show up (evdev/0/Nintendo Switch Left/Right Joy-Con) don't have any gyro or accelerometer fields attributed to them in Dolphin. I've been going crazy trying to find some documentation about why I'm only seeing the buttons and thumbsticks. I already had hid-nintendo installed on my system (v2) and I just installed v3.1 with dkms-hid-nintendo to make sure that it's not a problem of obsolete software. Can you think of anything I may be missing? 

I know this is off-topic, if you prefer you can email me. I didn't want to email you out of the blue.

v1993, 2021-02-26:
@hoopahmadness You must bind devices that end with "IMU" to motion. If they don't appear, check and debug your dkms module installation. <details><summary>Example profile (left joycon)</summary>

```ini
[Profile]
Device = evdev/0/Nintendo Switch Left Joy-Con
Buttons/A = `DPAD_RIGHT`
Buttons/B = `DPAD_UP`
Buttons/1 = `DPAD_LEFT`
Buttons/2 = `DPAD_DOWN`
Buttons/- = TR | SELECT
Buttons/+ = `TR2`
Buttons/Home = Z
IMUAccelerometer/Up = `evdev/0/Nintendo Switch Left Joy-Con IMU:Accel Z+`
IMUAccelerometer/Down = `evdev/0/Nintendo Switch Left Joy-Con IMU:Accel Z-`
IMUAccelerometer/Left = `evdev/0/Nintendo Switch Left Joy-Con IMU:Accel Y+`
IMUAccelerometer/Right = `evdev/0/Nintendo Switch Left Joy-Con IMU:Accel Y-`
IMUAccelerometer/Forward = `evdev/0/Nintendo Switch Left Joy-Con IMU:Accel X+`
IMUAccelerometer/Backward = `evdev/0/Nintendo Switch Left Joy-Con IMU:Accel X-`
IMUGyroscope/Dead Zone = 1.0000000000000000
IMUGyroscope/Pitch Up = `evdev/0/Nintendo Switch Left Joy-Con IMU:Gyro Y-`*0.858
IMUGyroscope/Pitch Down = `evdev/0/Nintendo Switch Left Joy-Con IMU:Gyro Y+`*0.858
IMUGyroscope/Roll Left = `evdev/0/Nintendo Switch Left Joy-Con IMU:Gyro X-`*0.858
IMUGyroscope/Roll Right = `evdev/0/Nintendo Switch Left Joy-Con IMU:Gyro X+`*0.858
IMUGyroscope/Yaw Left = `evdev/0/Nintendo Switch Left Joy-Con IMU:Gyro Z+`*0.858
IMUGyroscope/Yaw Right = `evdev/0/Nintendo Switch Left Joy-Con IMU:Gyro Z-`*0.858
IMUIR/Horizontal FOV = 42.000000000000000
IMUIR/Recenter = hold(Z, 2)
Rumble/Motor = Sine
D-Pad/Up = `Axis 1-`
D-Pad/Down = `Axis 1+`
D-Pad/Left = `Axis 0-`
D-Pad/Right = `Axis 0+`
```
</details>

hoopahmadness, 2021-02-26:
Thank you, that makes it very clear that I have some problem because I am not seeing the second imu device. I've never had to debug a kernel module before, I am having a hard time finding out where to start.

hoopahmadness, 2021-02-26:
`[ 3848.662838] nintendo 0005:057E:2007.0020: unknown main item tag 0x0
[ 3848.662911] nintendo 0005:057E:2007.0020: hidraw14: BLUETOOTH HID v80.01 Gamepad [Joy-Con (R)] on f8:ac:65:ec:ea:40
[ 3849.083533] nintendo 0005:057E:2007.0020: using factory cal for left stick
[ 3849.094772] nintendo 0005:057E:2007.0020: using factory cal for right stick
[ 3849.216026] nintendo 0005:057E:2007.0020: using user cal for IMU
[ 3849.548538] nintendo 0005:057E:2007.0020: controller MAC = 98:B6:E9:89:2C:5F
[ 3849.658742] input: Nintendo Switch Right Joy-Con as /devices/pci0000:00/0000:00:01.2/0000:01:00.0/0000:02:08.0/0000:06:00.3/usb3/3-5/3-5:1.0/bluetooth/hci0/hci0:512/0005:057E:2007.0020/input/input62
[ 3849.658865] input: Nintendo Switch Right Joy-Con IMU as /devices/pci0000:00/0000:00:01.2/0000:01:00.0/0000:02:08.0/0000:06:00.3/usb3/3-5/3-5:1.0/bluetooth/hci0/hci0:512/0005:057E:2007.0020/input/input63
[ 3849.867280] nintendo 0005:057E:2007.0020: compensating for 4 dropped IMU reports
[ 3849.867283] nintendo 0005:057E:2007.0020: delta=84 avg_delta=15`

According to dmesg It's seeing each joycon and assigning two devices, so that's good. I guess I just need to find out why the IMU devices aren't showing up in Dolphin

hoopahmadness, 2021-02-26:
Oh, it's a permissions issue. running dolphin with sudo shows the missing devices. I think I read that joycond includes a udev fix so I'll check that out. Thank you for your help!

DanielOgorchock, 2021-03-06:
It's very possible I have the math wrong for the gyros. I might be applying the calibration data incorrectly.