DanielOgorchock/linux #21: Some features remain unique to dkms-hid-nintendo

v1993, 2021-02-01:
As of right now, few PRs were merged into https://github.com/nicman23/dkms-hid-nintendo, but are not reflected by this repository. Those specifically include https://github.com/nicman23/dkms-hid-nintendo/pull/25, https://github.com/nicman23/dkms-hid-nintendo/pull/23 and not merged yet https://github.com/nicman23/dkms-hid-nintendo/pull/27.

DanielOgorchock, 2021-03-06:
Thanks for the heads up. I had intended to already have this driver merged into mainline kernel, but I've had little time to work on it lately (hopefully soon). I'm thinking the authors of those additional patches can submit to the mailing list at that point.

bitmeal, 2021-05-05:
@DanielOgorchock first a big thank you for all your work on the subject!

Should we not try to coordinate and merge the fixes/features to your repo and get a more complete version of the driver in the kernel? Your last entries on the mailing list are from 08.2020, if i saw this right. I suppose there has been no other activity for getting this merged and it would not pose that much of an issue to integrate the available fixes?

*(I am the author of one of the fixes/features mentioned above)*

DanielOgorchock, 2021-05-06:
@bitmeal Yeah we can bring in everything here, and I can resubmit to the mailing list with it all included. I'd be happy to accept some pull requests.

Earlier this year there was a bit more feedback on my latest submission to the list I need to address as well.