DanielOgorchock/linux #22: "Improve rumble performance and stability" commit seems to do the opposite

v1993, 2021-04-02:
This have been observed both by me and [some other people](https://github.com/nicman23/dkms-hid-nintendo/issues/33) that last change actually causes Bluetooth to drop connection more often.

v1993, 2021-04-28:
More testing reveals that issues happen on both versions, but they are different. Old method gives more disconnects but accel/gyro drifts less while new is less likely to disconnect but accel/gyro drifts more during rumble.

nicman23, 2021-04-28:
i am thinking of disabling support for it in the dkms package tbh.

nicman23, 2021-04-28:
i mean rumble it is not stable for bluetooth.

DanielOgorchock, 2021-05-06:
Yeah rumble over bluetooth hasn't been stable with a couple adapters I've tried. I do believe the patch is an improvement though. It actually results in less traffic from rumble and more closely reflects what the Switch does.

IIRC using the driver with the switchroot project on an actual console yielded more reliable results for me. I wonder if it's due to the bluetooth adapters in question.

DanielOgorchock, 2021-07-25:
I think my latest patchset improves BT rumble reliability by a large margin. Guess we'll have to wait and see as people start using the improvements