DanielOgorchock/linux #26: Is there a way to mapping the conrtoller it so A means A?

EviTRea, 2021-08-09:
So when I use the Kernel post 5.10, and doesn't have the Steam controller support on, this is what my machine use to recognize my Switch Pro controller right?

So an annoying thing is, I'm very used to Nintendo button layout, so when I'm playing, A means the right button for me.

But when I use the Linux controller support, it will take my A as B, B as A, as I've tried it in Enter the Gungeon binding.

If I use Steam's Pro Controller support, I can make it use the Nintendo Layout; but it's not really reliable since it sometimes load the profile too late and force me to restart the game.

If I can make it so the system would register my Input as it is, it would be so much easier for me. 

Is that a thing already? Or is this now a feature request?

(Side note: I use Godot Engine to make things sometimes. 
It doesn't work when using Steam input, probably because the compiled testing program didn't run through Steam;
when testing with Native Linux Pro Controller support, it works perfectly.
But in the Godot Input setting, the button is called "Dualshock Circle, Xbox B, Nintendo A",
Does that mean if the remapping works in all the other games, I could potentially revert it In Godot?
Not the big deal, but still kind of curious.)

nezd5553, 2021-08-17:
I was able to do this with this [environment variable from the Arch WIki](https://wiki.archlinux.org/title/Gamepad#Using_hid-nintendo_with_SDL2_Games). 
I swapped A/B and X/Y and put it in /etc/environment:
`SDL_GAMECONTROLLERCONFIG="050000007e0500000920000001800000,Nintendo Switch Pro Controller,platform:Linux,a:b1,b:b0,x:b2,y:b3,back:b9,guide:b11,start:b10,leftstick:b12,rightstick:b13,leftshoulder:b5,rightshoulder:b6,dpup:b14,dpdown:b15,dpleft:b16,dpright:b17,leftx:a0,lefty:a1,rightx:a2,righty:a3,lefttrigger:b7,righttrigger:b8,
030000007e0500000920000011810000,Nintendo Switch Pro Controller,platform:Linux,a:b1,b:b0,x:b2,y:b3,back:b9,guide:b11,start:b10,leftstick:b12,rightstick:b13,leftshoulder:b5,rightshoulder:b6,dpup:h0.1,dpdown:h0.4,dpleft:h0.8,dpright:h0.2,leftx:a0,lefty:a1,rightx:a2,righty:a3,lefttrigger:b7,righttrigger:b8,
060000007e0500000620000000000000,Nintendo Switch Combined Joy-Cons,platform:Linux,a:b1,b:b0,x:b2,y:b3,back:b9,guide:b11,start:b10,leftstick:b12,rightstick:b13,leftshoulder:b5,rightshoulder:b6,dpup:b14,dpdown:b15,dpleft:b16,dpright:b17,leftx:a0,lefty:a1,rightx:a2,righty:a3,lefttrigger:b7,righttrigger:b8,
"`

dogtopus, 2021-08-17:
This should be implemented on a game-to-game basis since it ultimately depends on the game to convert ev_msc scan codes to button prompts. If the game dev is ignorant or too big to listen for suggestions, maybe just write a small uinput script that swaps the buttons around.

SuperSamus, 2022-07-24:
SDL supports an environment variable named `SDL_GAMECONTROLLER_USE_BUTTON_LABELS` which, when set to 1, does what it says on the tin.
The Switch Pro Controller isn't supported yet, but it will be on the next version of SDL.

SuperSamus, 2022-07-24:
There also seems to be a kernel patch that addresses this.
https://patchwork.kernel.org/project/linux-input/patch/20220512001500.16739-2-fletcher0max@gmail.com/

SuperSamus, 2022-10-12:
With SDL 2.24.0 it should be like you are asking by default.
Set `SDL_GAMECONTROLLER_USE_BUTTON_LABELS=0` if you want to match Xbox's layout instead.

EviTRea, 2022-10-13:
> With SDL 2.24.0 it should be like you are asking by default. Set `SDL_GAMECONTROLLER_USE_BUTTON_LABELS=0` if you want to match Xbox's layout instead.

Just loaded up the same game (Enter the Gungeon) and played without Steam Switch settings, and the mapping is just working as expected.
Not sure if original issue was truly solved since a lot had changed since (such as Steam, kernel, DE and Distro: Pop!_OS with Cosmic to Fedora KDE), but I'm not experiencing anything related to mapping inverting at this point, so I'm closing this.

closed 2022-10-13 by EviTRea