DanielOgorchock/linux #27: [v15] Fails to compile with kernel 5.13(.14): ‘LED_FUNCTION_PLAYER’ undeclared here

thican, 2021-09-08:
Under current 5.13.14 kernel version, after applying also current hid-nintendo patches version v15 (from [HID: nintendo v15, Linux Input Mailing List](https://patchwork.kernel.org/project/linux-input/list/?series=540915)), module `drivers/hid/hid-nintendo.o` fails to compile with those error messages:
```make
  CALL    scripts/checksyscalls.sh
  CALL    scripts/atomic/check-atomics.sh
  DESCEND objtool
  CC [M]  drivers/hid/hid-nintendo.o
drivers/hid/hid-nintendo.c:406:9: error: ‘LED_FUNCTION_PLAYER’ undeclared here (not in a function); did you mean ‘LED_FUNCTION_POWER’?
  406 |         LED_FUNCTION_PLAYER "-1",
      |         ^~~~~~~~~~~~~~~~~~~
      |         LED_FUNCTION_POWER
drivers/hid/hid-nintendo.c:406:29: error: expected ‘}’ before string constant
  406 |         LED_FUNCTION_PLAYER "-1",
      |                             ^~~~
drivers/hid/hid-nintendo.c:405:55: note: to match this ‘{’
  405 | static const char * const joycon_player_led_names[] = {
      |                                                       ^
drivers/hid/hid-nintendo.c: In function ‘joycon_leds_create’:
drivers/hid/hid-nintendo.c:1880:58: error: expected ‘)’ before string constant
 1880 |                                       LED_FUNCTION_PLAYER "-5");
      |                                                          ^~~~~
      |                                                          )
drivers/hid/hid-nintendo.c:1877:38: note: to match this ‘(’
 1877 |                 name = devm_kasprintf(dev, GFP_KERNEL, "%s:%s:%s",
      |                                      ^
make[2]: *** [scripts/Makefile.build:273: drivers/hid/hid-nintendo.o] Error 1
make[1]: *** [scripts/Makefile.build:516: drivers/hid] Error 2
make: *** [Makefile:1862: drivers] Error 2
```

Steps to reproduce:
1. Fetch, uncompress and apply patches (dl as mbox from “Linux Input Mailing List”) to kernel `linux-5.13.14`;
2. Provide in kernel directory’s root `.config` file with option `CONFIG_HID_NINTENDO=m` (or =y, however not tested in my case);
3. Run `make modules_prepare` and `make drivers/hid/hid-nintendo.o` -- no jobs specified to avoid race condition. Full log for module compilation pasted above.

I can provide more information if required.

Thanks for support, best regards.

thican, 2021-09-11:
Hello,

I noticed this `LED_FUNCTION_PLAYER` is introduced by another patch from the Linux Input Mailing List, as the “HID: playstation: add LED support.” serie:
[[2/3] leds: add new LED_FUNCTION_PLAYER for player LEDs for game controllers.](https://patchwork.kernel.org/project/linux-input/patch/20210602061253.5747-3-roderick@gaikai.com/)

However, from the discussion in above link, this simple "LED_FUNCTION_PLAYER" has been dropped for a new patchset version which explicitly gives (only) five defined values, `LED_FUNCTION_PLAYER{1..5}`.

I guess a short-term solution is not to rely on the patch for "include/dt-bindings/leds/common.h" and simply to add `LED_FUNCTION_PLAYER` in your own patch; otherwise we will need to rewrite the loop using the `joycon_player_led_names` array.

Best regards,

thican, 2021-09-11:
Sorry, I forgot to mention the link to the new patchset serie:
[[v3] HID: playstation: add LED support](https://patchwork.kernel.org/project/linux-input/list/?series=543917)

thican, 2021-09-11:
Just to confirm, I just did a rebuild with now kernel version 5.13.15, simply adding this line just before the `joycon_player_led_names` array made it work.
```C
#define LED_FUNCTION_PLAYER "player"
```

DanielOgorchock, 2021-09-11:
Hi thican. I just submitted a new v16 of the patchset which depends on Roderick's latest iteration.  I am trying to keep things in sync with his latest patches to hopefully have hid-nintendo ready to go for merging once the led maintainer accepts the LED defines patch.

thican, 2021-09-11:
Thanks @DanielOgorchock for your work, I appreciate it.

Note for people currently trying on kernel 5.13, or maybe even 5.14: as noted in main message[0], it depends on the patches from the HID PlayStation series[1], which also need the patch for file `Documentation/leds/well-known-leds.txt`[2].

0: https://patchwork.kernel.org/project/linux-input/cover/20210911173639.5688-1-djogorchock@gmail.com/
1: https://patchwork.kernel.org/project/linux-input/list/?series=543917
2: https://github.com/DanielOgorchock/linux/commit/09f1273064eea23ec41fb206f6eccc2bf79d1fa1.patch

Sorry to bother you with this question, but since I got your attention I want to ask you this:
I am new to the process on integrating patches in upstream Linux repository, but is there a merge request (or something related) for your HID Nintendo series into upstream kernel? The purpose is to better follow, test, and even have a grasp on why your work is not(?) accepted yet.

Thanks again, and have a nice day.

PS: I was able to compile your module (and the whole kernel), however compilation fails for `hid-playstation` with just the patches from link 1; but it’s not in the scope of this issue and I am not (currently) interested in the support for this hardware, so it’s okay for me.

I guess we can close this issue.

thican, 2021-09-11:
Solved with v16.

closed 2021-09-11 by thican