DanielOgorchock/linux #28: Switch Pro Controller: X and Y buttons are swapped in joystick API

SuperSamus, 2021-11-08:
On my Switch Pro Controller (never updated the firmware) the mappings are correct with the evdev API, but not with the legacy joystick API.
The west button (X on Xbox, Y on Nintendo) is mapped as button 3, and the north button (Y on Xbox, X on Nintendo) is mapped as button 2. This should be the other way around. I never changed settings and I am not using joycond.

The A and B button are mapped correctly:  the south button (A on Xbox, B on Nintendo) is mapped as button 0, and the east button (B on Xbox, A on Nintendo) is mapped as button 1.

I am using NixOS unstable, with Linux 5.14.15-zen1 x86_64, and hid-nintendo 3.2 (rev 1c262xarsl).

SuperSamus, 2022-01-11:
I switched to the module from Linux 5.16, and even wiped /etc (NixOS regenerated it).
The bug still persists to me.

nicman23, 2022-01-12:
you can change / configure that from userspace. look at arch wiki, specifically on the nintendo switch controller section for a ready made string for your environment (meaning env).

SuperSamus, 2022-01-17:
It works. Thanks you!
But before closing, why isn't this correct by default?
If the Arch Wiki and the [SDL database](https://github.com/gabomdq/SDL_GameControllerDB/blob/master/gamecontrollerdb.txt#L1114) both remap the controller like this, does it mean that this wrong behavior is "intended"?

nicman23, 2022-01-17:
mostly to follow 360 maps as many people do not look at the buttons

SuperSamus, 2022-01-17:
But it doesn't make much sense.
If I pretend that I'm holding an Xbox controller, A and B behave like I expect, while X and Y do not.
If instead I'd be looking at the controller and didn't know that Nintendo is different*, X and Y would behave like I expect, while A and B would not.
I don't understand...

(*Technically it's Microsoft who swapped the buttons, but this doesn't have anything to do with the discussion)

SeongGino, 2022-05-14:
Yeah, not gonna lie; if this has taken me several months just to find this issue and fix these two buttons mixed up on a Pro Controller, I feel like this is either an oversight or bad design in the module, and should be changed.

After all, isn't the idea to make it so that these devices can work out-of-the-box as smoothly as possible? If Linus Sebastian knew about this, he'd be livid.

(Sorry if this is necro-ing, but I felt it appropriate to complain about an issue that technically still persists in-kernel)

SuperSamus, 2022-05-14:
Maybe I should reopen this.

nicman23, 2022-05-14:
pull requests for this are welcome because i am not really available atm

nezd5553, 2022-05-14:
I think this is a problem with how controllers work through HID. In [input-event-codes.h](https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/tree/include/uapi/linux/input-event-codes.h#n382), A and B follow the Xbox layout, but X and Y are swapped.

```
...
#define BTN_SOUTH		0x130
#define BTN_A			BTN_SOUTH
#define BTN_EAST		0x131
#define BTN_B			BTN_EAST
#define BTN_C			0x132
#define BTN_NORTH		0x133
#define BTN_X			BTN_NORTH
#define BTN_WEST		0x134
#define BTN_Y			BTN_WEST
...
```
I tested an Xbox-style controller with `evtest` and it also showed X as north and Y as west.

SuperSamus, 2022-05-14:
On the topic of codes that don't match, I found out that IMU codes (for motion sensors) are also different between the Switch Pro and the Dualshock/Dualsense.
https://wiki.dolphin-emu.org/index.php?title=Motion_evdev
Maybe I should open another issue for that.

SuperSamus, 2022-05-14:
I tried to research a bit.
So, there are two interfaces: the new `evdev` interface, and the old `joystick` interface.
I have no idea whether drivers communicate with them separately, or if they are something that works by itself (I'm supposing the former, because if it's the latter, then what I'm saying next will be partially wrong).

On `evdev`, buttons are identified with codes. Documentation: https://www.kernel.org/doc/html/latest/input/gamepad.html.
It seems that on `evdev` the buttons for the Switch controller are mapped correctly, so it looks like Xbox is in the wrong there.

On `joystick`, buttons are only identified with a number. I couldn't find documentation on how they should be mapped ([this](https://www.kernel.org/doc/html/latest/input/joydev/joystick-api.html#js-event-number) is not clear at all), but it seems that from 0-3 they are mapped in order: south, east, west, north.
**If I'm right, the problem (that is the reason for the name of the issue) is that in the Pro Controller west is mapped as 3, and north is mapped as 2, instead of the opposite.**

Finally, what games use what. The [Arch Wiki](https://wiki.archlinux.org/title/Gamepad) says that `evdev` is basically the only thing being used, even SDL is using it. I did a bit of research, and it seems that the wiki is **wrong**.
First, SDL. Basically everything uses SDL, otherwise the workaround I described above definitely wouldn't work. And while SDL does use `evdev`, it's not used for gamepads: for them, it uses good ol' `joystick`. The evidence:

- [SDL_gamecontroller.c](https://github.com/libsdl-org/SDL/blob/main/src/joystick/SDL_gamecontroller.c) imports `SDL_joystick_c.h` and doesn't import `SDL_evdev.h`.
- The comments on [SDL_evdev.c](https://github.com/libsdl-org/SDL/blob/main/src/core/linux/SDL_evdev.c) refer to mice/touchpads/etc., but never to gamepads;
- The remapping syntax  (like the one on the workaround) uses numbers (like `joystick`) to describe the pressed button, instead of "words" (like `evdev`);

So, that's basically everything I wanted to say...

...except that after I wrote this wall of text, I found out that SDL [has a controller database](https://github.com/libsdl-org/SDL/blob/main/src/joystick/SDL_gamecontrollerdb.h). So... what is happening? Is it not being used? Is it SDL fault or is it whatever application I tried? (Mostly Wine, of course.)

So... I don't know what to add.

SuperSamus, 2022-05-30:
I see that this is being tackled here: https://www.spinics.net/lists/linux-input/msg78346.html.

SeongGino, 2022-05-30:
Said patch seems to have been stalled for a few days, though.

SuperSamus, 2022-10-12:
Fixed with SDL 2.24.0.
(By default, SDL sets `SDL_GAMECONTROLLER_USE_BUTTON_LABELS=1`, which swaps both A/B and X/Y to match the button labels.
Set `SDL_GAMECONTROLLER_USE_BUTTON_LABELS=0` to match the positional layout.
Discussion about this on https://github.com/libsdl-org/SDL/issues/6117)

closed 2022-10-12 by SuperSamus