DanielOgorchock/linux #29: Pro Controller y-axes and drift problem

slobodan-rs, 2021-11-13:
Hello,

I am having an odd problem with hid driver. The controller is working fine wired but on bluetooth I have a "calibration" problem. Immediately upon connecting the calibration with sticks is fine but if you press any other button on controller, other then sticks it messes up axes so "UP" doesn't go all the way and center drifts (the video explains it better). The numbers in calibration tool goes all over the place.

Also it loses connection every 5 min, have been testing it to find out pattern why that happens.    

Arch Linux with kernel 5.15.2, Hid-Nintendo 3.2 Try it also on different distros and linux kernels the problem persists..

[https://reddit.com/link/qrwdke/video/9chgk9mui1z71/player](https://reddit.com/link/qrwdke/video/9chgk9mui1z71/player)

Thank you!

TheZipBon, 2022-02-13:
I am also experiencing this issue.

TheZipBon, 2022-02-13:
I have the same issue with the analog sticks on Linux Mint with 5.13 kernel.

slobodan-rs, 2022-02-22:
from my experience it's fix in 5.15 kernel