DanielOgorchock/linux #30: System freeze after connecting 2 joy cons (Nintendo Switch)

jagiella, 2021-11-20:
My system is an ubuntu 22.04 jammy jellyfish (developement release) with gnome desktop.

I tried the hid-nintendo in two setups so far:
1. mainline kernel 5.15 with dkms-hid-nintendo as kernel extension (from [nicman23s repository](https://github.com/nicman23/dkms-hid-nintendo))
2. mainline kernel 5.16-rc1 with hid-nintendo built into the kernel
To my knowledge both drivers are based on or directly derived from your code.

I can pair and connect the joy cons via bluetooth. Unfortunately after having 2 joy cons (left and right) connected the system will eventually freeze the desktop. The mouse cursor is still moving, but keyboard is unresponsive (even changing tty with ctrl+alt+F1...F8) and everything is frozen. I am forced to unplug power to shutdown/reboot the system.

By the way, I tried that before with 8bitdo sf30 pro controller in switch mode and its successfully contecting as switch pro controller and not hanging up. But with original joy cons from Nintendo it persistently freezes.

Also installed the joycond (and joycond-cemuhook).

As the issue even persists in the new mainline kernel 5.16-rc1 I felt the need to warn you about this issue as it might cause problems for others as well in the future using that kernel.

jagiella, 2021-11-20:
Okay I tried to reproduce the issue and this time it did not freeze, but the bluetooth ended up not being usable anymore.

**What did I do?**
- The 2 joy cons were already paired from a previous session.
- I switched both joy cons on.
- Both connected: LEDs showed joycon ids (left=1 LED, right=2 LEDs) and the bluetooth settings of ubuntu showed both as connected.

**Whats the issue?**
- None of the controllers were usable / visible as input devices.
- Deleting the joycons (for the purpose of repairing) didn't work anymore.
- The whole bluetooth setting menu is not working anymore:
  - impossible to disable or enable bluetooth 
  - impossible to remove the joy cons
- rebooting from commandline (sudo reboot) leads to a frozen system as initially discribed in the ticket. Needs a hardware reset.  

<details>
<summary>**kernel log (sudo journalctl -k -b -1)**</summary>

```
-- Journal begins at Sat 2021-11-06 20:42:51 CET, ends at Sat 2021-11-20 15:08:44 CET. --
Nov 20 14:52:05 nick-G501JW kernel: microcode: microcode updated early to revision 0x28, date = 2019-11-12
Nov 20 14:52:05 nick-G501JW kernel: Linux version 5.16.0-051600rc1-generic (kernel@kathleen) (gcc (Ubuntu 11.2.0-10ubuntu1) 11.2.0, GNU ld (GNU Binutils for Ubuntu) 2.37) #202111142330 SMP PREEMPT Sun Nov 14 23:35:10 UTC 2021
Nov 20 14:52:05 nick-G501JW kernel: Command line: BOOT_IMAGE=/boot/vmlinuz-5.16.0-051600rc1-generic root=UUID=89c7edd7-4bc5-4664-97d5-947d91d5a4cf ro quiet splash vt.handoff=7
Nov 20 14:52:05 nick-G501JW kernel: KERNEL supported cpus:
Nov 20 14:52:05 nick-G501JW kernel:   Intel GenuineIntel
Nov 20 14:52:05 nick-G501JW kernel:   AMD AuthenticAMD
Nov 20 14:52:05 nick-G501JW kernel:   Hygon HygonGenuine
Nov 20 14:52:05 nick-G501JW kernel:   Centaur CentaurHauls
Nov 20 14:52:05 nick-G501JW kernel:   zhaoxin   Shanghai  
Nov 20 14:52:05 nick-G501JW kernel: x86/fpu: Supporting XSAVE feature 0x001: 'x87 floating point registers'
Nov 20 14:52:05 nick-G501JW kernel: x86/fpu: Supporting XSAVE feature 0x002: 'SSE registers'
Nov 20 14:52:05 nick-G501JW kernel: x86/fpu: Supporting XSAVE feature 0x004: 'AVX registers'
Nov 20 14:52:05 nick-G501JW kernel: x86/fpu: xstate_offset[2]:  576, xstate_sizes[2]:  256
Nov 20 14:52:05 nick-G501JW kernel: x86/fpu: Enabled xstate features 0x7, context size is 832 bytes, using 'standard' format.
Nov 20 14:52:05 nick-G501JW kernel: signal: max sigframe size: 1776
Nov 20 14:52:05 nick-G501JW kernel: BIOS-provided physical RAM map:
Nov 20 14:52:05 nick-G501JW kernel: BIOS-e820: [mem 0x0000000000000000-0x0000000000057fff] usable
Nov 20 14:52:05 nick-G501JW kernel: BIOS-e820: [mem 0x0000000000058000-0x0000000000058fff] reserved
Nov 20 14:52:05 nick-G501JW kernel: BIOS-e820: [mem 0x0000000000059000-0x000000000009dfff] usable
Nov 20 14:52:05 nick-G501JW kernel: BIOS-e820: [mem 0x000000000009e000-0x000000000009ffff] reserved
Nov 20 14:52:05 nick-G501JW kernel: BIOS-e820: [mem 0x0000000000100000-0x00000000728e9fff] usable
Nov 20 14:52:05 nick-G501JW kernel: BIOS-e820: [mem 0x00000000728ea000-0x00000000728f0fff] ACPI NVS
Nov 20 14:52:05 nick-G501JW kernel: BIOS-e820: [mem 0x00000000728f1000-0x000000007321afff] usable
Nov 20 14:52:05 nick-G501JW kernel: BIOS-e820: [mem 0x000000007321b000-0x00000000734b8fff] reserved
Nov 20 14:52:05 nick-G501JW kernel: BIOS-e820: [mem 0x00000000734b9000-0x000000007a012fff] usable
Nov 20 14:52:05 nick-G501JW kernel: BIOS-e820: [mem 0x000000007a013000-0x000000007a209fff] reserved
Nov 20 14:52:05 nick-G501JW kernel: BIOS-e820: [mem 0x000000007a20a000-0x000000007a3f7fff] usable
Nov 20 14:52:05 nick-G501JW kernel: BIOS-e820: [mem 0x000000007a3f8000-0x000000007ab00fff] ACPI NVS
Nov 20 14:52:05 nick-G501JW kernel: BIOS-e820: [mem 0x000000007ab01000-0x000000007affefff] reserved
Nov 20 14:52:05 nick-G501JW kernel: BIOS-e820: [mem 0x000000007afff000-0x000000007affffff] usable
Nov 20 14:52:05 nick-G501JW kernel: BIOS-e820: [mem 0x000000007bc00000-0x000000007fdfffff] reserved
Nov 20 14:52:05 nick-G501JW kernel: BIOS-e820: [mem 0x00000000f0000000-0x00000000f7ffffff] reserved
Nov 20 14:52:05 nick-G501JW kernel: BIOS-e820: [mem 0x00000000fec00000-0x00000000fec00fff] reserved
Nov 20 14:52:05 nick-G501JW kernel: BIOS-e820: [mem 0x00000000fed00000-0x00000000fed03fff] reserved
Nov 20 14:52:05 nick-G501JW kernel: BIOS-e820: [mem 0x00000000fed1c000-0x00000000fed1ffff] reserved
Nov 20 14:52:05 nick-G501JW kernel: BIOS-e820: [mem 0x00000000fee00000-0x00000000fee00fff] reserved
Nov 20 14:52:05 nick-G501JW kernel: BIOS-e820: [mem 0x00000000ff000000-0x00000000ffffffff] reserved
Nov 20 14:52:05 nick-G501JW kernel: BIOS-e820: [mem 0x0000000100000000-0x000000047f1fffff] usable
Nov 20 14:52:05 nick-G501JW kernel: NX (Execute Disable) protection: active
Nov 20 14:52:05 nick-G501JW kernel: e820: update [mem 0x724a1018-0x724cb857] usable ==> usable
Nov 20 14:52:05 nick-G501JW kernel: e820: update [mem 0x724a1018-0x724cb857] usable ==> usable
Nov 20 14:52:05 nick-G501JW kernel: extended physical RAM map:
Nov 20 14:52:05 nick-G501JW kernel: reserve setup_data: [mem 0x0000000000000000-0x0000000000057fff] usable
Nov 20 14:52:05 nick-G501JW kernel: reserve setup_data: [mem 0x0000000000058000-0x0000000000058fff] reserved
Nov 20 14:52:05 nick-G501JW kernel: reserve setup_data: [mem 0x0000000000059000-0x000000000009dfff] usable
Nov 20 14:52:05 nick-G501JW kernel: reserve setup_data: [mem 0x000000000009e000-0x000000000009ffff] reserved
[... cut ...]
Nov 20 14:58:29 nick-G501JW kernel: ntfs3: Read-only LZX/Xpress compression included
Nov 20 14:58:29 nick-G501JW kernel: ntfs3: Unknown parameter 'windows_names'
Nov 20 15:01:01 nick-G501JW kernel: Bluetooth: hci0: command 0x0804 tx timeout
Nov 20 15:01:01 nick-G501JW kernel: Bluetooth: hci0: failed to restart LE scan: status 31
Nov 20 15:01:03 nick-G501JW kernel: Bluetooth: hci0: failed to disable LE scan: status 0x1f
Nov 20 15:02:07 nick-G501JW kernel: Bluetooth: hci0: HCI reset during shutdown failed
```
</details>

CactiChameleon9, 2022-04-11:
I have this issue too! However, for me a single joycon is the problem (my left one) and the right one connects fine as normal (if connected first). Do you always connect them in the same order?

My system sometimes crashes immediately or sometimes has a bit of a wait until things break (enough time to connect the good controller)

Here are 2 dmesgs of the issue:
```
[    0.000000] Linux version 5.16.16-xm1.0e20220113.fc34.x86_64 (mockbuild@f6a3917b366f445f93e2d36b2ee49117) (clang version 12.0.1 (Fedora 12.0.1-1.fc34), LLD 12.0.1) #1 SMP Mon Mar 21 12:15:09 UTC 2022
[    0.000000] Command line: BOOT_IMAGE=(hd0,gpt2)/vmlinuz-5.16.16-xm1.0e20220113.fc34.x86_64 root=/dev/mapper/linux-root ro rd.lvm.lv=linux/root rd.luks.uuid=luks-bec97561-22e4-4df2-a449-5639a24bbff2 rhgb quiet psi=1 video=eDP-1:1920x1080@40
[    0.000000] KERNEL supported cpus:
[    0.000000]   Intel GenuineIntel
[    0.000000]   AMD AuthenticAMD
[    0.000000]   Hygon HygonGenuine
[    0.000000]   Centaur CentaurHauls
[    0.000000]   zhaoxin   Shanghai  
[    0.000000] x86/fpu: Supporting XSAVE feature 0x001: 'x87 floating point registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x002: 'SSE registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x004: 'AVX registers'
[    0.000000] x86/fpu: xstate_offset[2]:  576, xstate_sizes[2]:  256
[    0.000000] x86/fpu: Enabled xstate features 0x7, context size is 832 bytes, using 'compacted' format.
[    0.000000] signal: max sigframe size: 1776
[    0.000000] BIOS-provided physical RAM map:
[    0.000000] BIOS-e820: [mem 0x0000000000000000-0x000000000009efff] usable
[    0.000000] BIOS-e820: [mem 0x000000000009f000-0x00000000000bffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000000100000-0x0000000009afffff] usable
[    0.000000] BIOS-e820: [mem 0x0000000009b00000-0x0000000009dfffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000009e00000-0x0000000009efffff] usable
[    0.000000] BIOS-e820: [mem 0x0000000009f00000-0x0000000009f0ffff] ACPI NVS
[    0.000000] BIOS-e820: [mem 0x0000000009f10000-0x00000000bab68fff] usable
[    0.000000] BIOS-e820: [mem 0x00000000bab69000-0x00000000bbd68fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000bbd69000-0x00000000c9f7efff] usable
[    0.000000] BIOS-e820: [mem 0x00000000c9f7f000-0x00000000ccf7efff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000ccf7f000-0x00000000cdf7efff] ACPI NVS
[    0.000000] BIOS-e820: [mem 0x00000000cdf7f000-0x00000000cdffefff] ACPI data
[    0.000000] BIOS-e820: [mem 0x00000000cdfff000-0x00000000cdffffff] usable
[    0.000000] BIOS-e820: [mem 0x00000000ce000000-0x00000000ceffffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000f8000000-0x00000000fbffffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fde00000-0x00000000fdefffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fec00000-0x00000000fec00fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fec10000-0x00000000fec10fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fec20000-0x00000000fec20fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fed80000-0x00000000fed80fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fee00000-0x00000000fee00fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000ff000000-0x00000000fff1ffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000100000000-0x000000020f33ffff] usable
[    0.000000] BIOS-e820: [mem 0x000000020f340000-0x000000022fffffff] reserved
[    0.000000] NX (Execute Disable) protection: active
[    0.000000] e820: update [mem 0xb6c0e018-0xb6c1b457] usable ==> usable
[    0.000000] e820: update [mem 0xb6c0e018-0xb6c1b457] usable ==> usable
[    0.000000] extended physical RAM map:
[    0.000000] reserve setup_data: [mem 0x0000000000000000-0x000000000009efff] usable
[    0.000000] reserve setup_data: [mem 0x000000000009f000-0x00000000000bffff] reserved
[    0.000000] reserve setup_data: [mem 0x0000000000100000-0x0000000009afffff] usable
[    0.000000] reserve setup_data: [mem 0x0000000009b00000-0x0000000009dfffff] reserved
[    0.000000] reserve setup_data: [mem 0x0000000009e00000-0x0000000009efffff] usable
[    0.000000] reserve setup_data: [mem 0x0000000009f00000-0x0000000009f0ffff] ACPI NVS
[    0.000000] reserve setup_data: [mem 0x0000000009f10000-0x00000000b6c0e017] usable
[    0.000000] reserve setup_data: [mem 0x00000000b6c0e018-0x00000000b6c1b457] usable
[    0.000000] reserve setup_data: [mem 0x00000000b6c1b458-0x00000000bab68fff] usable
[    0.000000] reserve setup_data: [mem 0x00000000bab69000-0x00000000bbd68fff] reserved
[    0.000000] reserve setup_data: [mem 0x00000000bbd69000-0x00000000c9f7efff] usable
[    0.000000] reserve setup_data: [mem 0x00000000c9f7f000-0x00000000ccf7efff] reserved
[    0.000000] reserve setup_data: [mem 0x00000000ccf7f000-0x00000000cdf7efff] ACPI NVS
[    0.000000] reserve setup_data: [mem 0x00000000cdf7f000-0x00000000cdffefff] ACPI data
[    0.000000] reserve setup_data: [mem 0x00000000cdfff000-0x00000000cdffffff] usable
[    0.000000] reserve setup_data: [mem 0x00000000ce000000-0x00000000ceffffff] reserved
[    0.000000] reserve setup_data: [mem 0x00000000f8000000-0x00000000fbffffff] reserved
[    0.000000] reserve setup_data: [mem 0x00000000fde00000-0x00000000fdefffff] reserved
[    0.000000] reserve setup_data: [mem 0x00000000fec00000-0x00000000fec00fff] reserved
[    0.000000] reserve setup_data: [mem 0x00000000fec10000-0x00000000fec10fff] reserved
[    0.000000] reserve setup_data: [mem 0x00000000fec20000-0x00000000fec20fff] reserved
[... cut ...]
[  160.442830] Code: ff ff ff 85 c0 79 9b 49 c7 c4 ff ff ff ff 5b 5d 4c 89 e0 41 5c c3 66 0f 1f 84 00 00 00 00 00 f3 0f 1e fa b8 10 00 00 00 0f 05 <48> 3d 01 f0 ff ff 73 01 c3 48 8b 0d dd 2a 0f 00 f7 d8 64 89 01 48
[  160.442831] RSP: 002b:00007fff82054008 EFLAGS: 00000246 ORIG_RAX: 0000000000000010
[  160.442833] RAX: ffffffffffffffda RBX: 00007fff82054040 RCX: 00007fa62b73031b
[  160.442834] RDX: 00007fff82054040 RSI: 0000000040086409 RDI: 000000000000000e
[  160.442835] RBP: 0000000040086409 R08: 0000559fe80fd520 R09: 0000000000000000
[  160.442835] R10: 0000000000000001 R11: 0000000000000246 R12: 0000559fe51143f8
[  160.442836] R13: 000000000000000

CactiChameleon9, 2022-04-12:
I managed to fix the issue on my end. Not too sure how this fixed it, but I switched back to fedora's main kernal from xanmod's kernel. Hope this helps someone

arieleoar, 2022-11-29:
Hi Everyone!

I confirm that this is also happening in Ubuntu 22.10, connecting any of the joycons to the pc bluetooth makes the system crash, is there a workaround for this? or a way we could help to fix it?

I'm attaching the kernel crashes:
[Nov 29 09 30 17 Kernel Log.txt](https://github.com/DanielOgorchock/linux/files/10113328/Nov.29.09.30.17.Kernel.Log.txt)
[Nov 29 09 32 15 Kernel Log.txt](https://github.com/DanielOgorchock/linux/files/10113330/Nov.29.09.32.15.Kernel.Log.txt)