DanielOgorchock/linux #34: hid-nintendo module fails to load

Zixiken, 2022-02-02:
OS: Linux Mint 20 (also updated to 20.3, but they have the same kernel)
Kernel: 5.13.0-28-generic

dkms operations work with no errors, but modprobe/insmod fails with: "ERROR: could not insert 'hid_nintendo': Exec format error". dmesg/journalctl output additionally shows: "module: x86/modules: Skipping invalid relocation target, existing value is nonzero for type 1, loc 0000000085ff9d82, val ffffffffc148a694".

Zixiken, 2022-02-02:
Just confirmed that my previous kernel (5.8.0-63-generic) still works just fine with the same module source code.

nicman23, 2022-02-03:
> hid_nintendo: Exec format error

 did you do anything weird compiling it? this usually mean wrong artitecture

Zixiken, 2022-02-03:
No, just the commands in the readme. I did some searching and saw people getting errors like this with other kernel modules relating to the version of gcc used...does this mean the gcc version used to compile the kernel has to match what I used for the module? Is there a way for me to tell what version the packagers used?

nicman23, 2022-02-08:
yeah it is on the first line of dmesg

Zixiken, 2022-03-02:
Updated to kernel 5.13.0-30-generic when it was released, and I think the issue is fixed.

Thanks for that dmesg info though, I'll keep that in mind for the future.

closed 2022-03-02 by Zixiken