DanielOgorchock/linux #36: Gulikit King Kong Controller hang

pickfire, 2022-05-03:
I believe this is merged into linux but opening an issue here since it was mentioned in https://github.com/DanielOgorchock/joycond

I am using gulikit king kong controller with nintendo switch mode, but when I connect it to my computer, it freezes. There was once I upgraded the kernel (linux-zen) without reboot and it seemed to work fine, I guess some modules wasn't modprobed then it will work fine (no freeze) and detected a pro controller in yuzu but I am not sure which module.

```
5月 03 21:43:08 mi kernel: hid-generic 0005:045E:02E0.0006: unknown main item tag 0x0
5月 03 21:43:08 mi kernel: input: Pro Controller as /devices/pci0000:00/0000:00:08.1/0000:02:00.3/usb1/1-4/1-4:1.0/bluetooth/hci0/hci0:1/0005:045E:02E0.0006/input/input21
5月 03 21:43:08 mi kernel: hid-generic 0005:045E:02E0.0006: input,hidraw5: BLUETOOTH HID v9.03 Gamepad [Pro Controller] on 80:30:49:10:6b:d0
```

I got that log when I connect it through bluetooth, but on wired the computer freeze and keyboard caps lock led kept blinking.

Not sure if this is related but I got quite a bit of it in the journalctl logs.
```
Failed to find catalog entry: Invalid argument
```

`uname -a`

```
Linux mi 5.17.5-zen1-1-zen #1 ZEN SMP PREEMPT Wed, 27 Apr 2022 20:56:14 +0000 x86_64 GNU/Linux
```

I am not quite sure how to figure out what causes the computer to freeze but it happens after I changed the mode to switch controller, IIRC last time I saw something related to `failed to probe ...` mentioning `nintendo` in the journalctl but I can't it anymore.

dogtopus, 2022-05-03:
If you can, could you please use kdump to create a memory dump and at least post the backtrace here? This will save tremendous amount of time that will otherwise be wasted on trying to reproduce the kernel panic.

pickfire, 2022-05-04:
I tried doing a kdump following https://wiki.archlinux.org/title/Kdump and https://www.kernel.org/doc/html/latest/admin-guide/kdump/kdump.html but I still can't seemed to get the crash dump after spending hours on it, when it crash, the caps lock led is no longer blinking but I still don't see the crash dump file in the specified location, I also can't find the logs on why this happened.

dogtopus, 2022-05-04:
Did you try the `kdump-save.service` mentioned on the arch wiki?

Maybe also try switching to a console before you attempt to crash the system.

pickfire, 2022-05-05:
> Did you try the kdump-save.service mentioned on the arch wiki?

Yes, I put that within the initramfs using mkinitfs but I don't seemed to find it in the journal log when it crash, not sure why.

> Maybe also try switching to a console before you attempt to crash the system.

I also try crashing it in TTY2, and it also stuck the same way it did with X11, it did not go to the new console started from kexec, it just freezes.

dogtopus, 2022-05-05:
Does sysrq trigger work? I guess something might be wrong when kexec()ing and made it freeze

pickfire, 2022-05-06:
> Does crashing with sysrq trigger trigger the kdump properly? I guess something might went wrong during the crash kexec() and made it freeze.

No, I did `echo c > /proc/sysrq-trigger`, and also try crashing the system by turning on the controller on nintendo switch mode.

> Maybe you could also try other methods like getting a backtrace via the serial port or pstore dump. If everything fails you might need to run the kernel in a VM and attach a debugger.

But I don't know how to use serial port or pstore dump. Hmm, the last probably sounds as hard.

Is there an easier way like say I get to use a live rescue disk that I can easily plug on my laptop and crash it there to get the kdump? That way I can just run the live rescue disk and after it crash save the file to my disk or somewhere then I can send it here.

nicman23, 2022-05-12:
sorry i really cannot help with this. this is a downstream project (dkms)

pickfire, 2022-05-21:
Now after I updated kernel to `5.17.9-zen1-1-zen` it seemed like I no longer get kernel panic. But my computer still hangs.

<details>
<summary>Kernel log</summary>

```
...
5月 21 10:48:29 mi kernel: hid-generic 0005:057E:2009.0006: unknown main item tag 0x0
5月 21 10:48:29 mi kernel: input: Pro Controller as /devices/pci0000:00/0000:00:08.1/0000:02:00.3/usb1/1-4/1-4:1.0/bluetooth/hci0/hci0:3/0005:057E:2009.0006/input/input21
5月 21 10:48:29 mi kernel: hid-generic 0005:057E:2009.0006: input,hidraw5: BLUETOOTH HID v0.01 Gamepad [Pro Controller] on 80:30:49:10:6b:d0
5月 21 10:48:29 mi kernel: nintendo 0005:057E:2009.0006: unknown main item tag 0x0
5月 21 10:48:29 mi kernel: nintendo 0005:057E:2009.0006: hidraw5: BLUETOOTH HID v80.01 Gamepad [Pro Controller] on 80:30:49:10:6b:d0
5月 21 10:48:31 mi kernel: nintendo 0005:057E:2009.0006: using user cal for left stick
5月 21 10:48:31 mi kernel: nintendo 0005:057E:2009.0006: using user cal for right stick
5月 21 10:48:31 mi kernel: nintendo 0005:057E:2009.0006: using factory cal for IMU
5月 21 10:48:31 mi kernel: nintendo 0005:057E:2009.0006: controller MAC = 98:B6:E9:EC:11:08
5月 21 10:48:31 mi kernel: input: Nintendo Switch Pro Controller as /devices/pci0000:00/0000:00:08.1/0000:02:00.3/usb1/1-4/1-4:1.0/bluetooth/hci0/hci0:3/0005:057E:2009.0006/input/input22
5月 21 10:48:31 mi kernel: input: Nintendo Switch Pro Controller IMU as /devices/pci0000:00/0000:00:08.1/0000:02:00.3/usb1/1-4/1-4:1.0/bluetooth/hci0/hci0:3/0005:057E:2009.0006/input/input23
5月 21 10:48:31 mi kernel: divide error: 0000 [#1] PREEMPT SMP NOPTI
5月 21 10:48:31 mi kernel: CPU: 5 PID: 33678 Comm: khidpd_057e2009 Not tainted 5.17.9-zen1-1-zen #1 b447c96d85e73a9127528593906685354a7b7c5d
5月 21 10:48:31 mi kernel: Hardware name: TIMI RedmiBook 14 II/TM1951, BIOS RMARN4B0P0909 08/19/2020
5月 21 10:48:31 mi kernel: RIP: 0010:nintendo_hid_event+0x5ff/0xd20 [hid_nintendo]
5月 21 10:48:31 mi kernel: Code: 08 00 00 44 8b 83 04 08 00 00 89 c6 0f b7 04 24 8b bb 00 08 00 00 39 c8 0f 8e 51 04 00 00 29 c8 29 cf 89 c2 c1 e0 0f 29 d0 99 <f7> ff 44 8b 8b 10 08 00 00 8b bb 0c 08 00 00 ba 01 80 ff ff 0f b7
5月 21 10:48:31 mi kernel: RSP: 0018:ffffae304aee3d48 EFLAGS: 00010212
5月 21 10:48:31 mi kernel: RAX: 000000000403f7f8 RBX: ffff8ca346ea8028 RCX: 0000000000000000
5月 21 10:48:31 mi kernel: RDX: 0000000000000000 RSI: 0000000000000807 RDI: 0000000000000000
5月 21 10:48:31 mi kernel: RBP: ffff8ca35fd28184 R08: 0000000000000000 R09: 0000000000000001
5月 21 10:48:31 mi kernel: R10: 0000000000000001 R11: 0000000000000008 R12: ffff8ca346ea8790
5月 21 10:48:31 mi kernel: R13: 0000000000008000 R14: ffff8ca33bac1800 R15: ffff8ca346ea876c
5月 21 10:48:31 mi kernel: FS:  0000000000000000(0000) GS:ffff8ca43f740000(0000) knlGS:0000000000000000
5月 21 10:48:31 mi kernel: CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
5月 21 10:48:31 mi kernel: CR2: 00007fb0dbc47524 CR3: 000000039b610000 CR4: 0000000000350ee0
5月 21 10:48:31 mi kernel: Call Trace:
5月 21 10:48:31 mi kernel:  <TASK>
5月 21 10:48:31 mi kernel:  ? schedule+0x4b/0xc0
5月 21 10:48:31 mi kernel:  hid_input_report+0x127/0x160
5月 21 10:48:31 mi kernel:  hidp_session_thread+0x4ff/0x9c0 [hidp d07bb5eb2176277bd5f206374ffe91a88aa9a489]
5月 21 10:48:31 mi kernel:  ? hidp_close+0x10/0x10 [hidp d07bb5eb2176277bd5f206374ffe91a88aa9a489]
5月 21 10:48:31 mi kernel:  ? hidp_close+0x10/0x10 [hidp d07bb5eb2176277bd5f206374ffe91a88aa9a489]
5月 21 10:48:31 mi kernel:  ? do_wait_intr_irq+0xa0/0xa0
5月 21 10:48:31 mi kernel:  ? hidp_session_remove+0x180/0x180 [hidp d07bb5eb2176277bd5f206374ffe91a88aa9a489]
5月 21 10:48:31 mi kernel:  kthread+0x138/0x160
5月 21 10:48:31 mi kernel:  ? kthread_complete_and_exit+0x20/0x20
5月 21 10:48:31 mi kernel:  ret_from_fork+0x1f/0x30
5月 21 10:48:31 mi kernel:  </TASK>
5月 21 10:48:31 mi kernel: Modules linked in: hid_nintendo ff_memless hidp snd_seq_dummy snd_hrtimer snd_seq snd_seq_device ccm rfcomm tun cmac algif_hash algif_skcipher af_alg bnep nls_iso8859_1 vfat fat hid_apple btusb btrtl btbcm btintel btmtk bluetooth ecdh_generic crc16 usbhid apple_mfi_fastcharge intel_rapl_msr intel_rapl_common edac_mce_amd joydev kvm_amd snd_acp3x_pdm_dma mousedev snd_acp3x_rn kvm snd_soc_dmic irqbypass hid_multitouch serio_raw wmi_bmof crct10dif_pclmul snd_sof_amd_renoir lzo_rle crc32_pclmul snd_acp_config snd_sof_amd_acp rtw88_8822ce ghash_clmulni_intel snd_sof_pci rtw88_8822c aesni_intel crypto_simd snd_sof cryptd snd_hda_codec_realtek rapl snd_soc_acpi rtw88_pci snd_hda_codec_generic ledtrig_audio snd_hda_codec_hdmi pcspkr rtw88_core snd_hda_intel snd_soc_core snd_intel_dspcfg snd_compress snd_intel_sdw_acpi ac97_bus mac80211 snd_pcm_dmaengine snd_hda_codec snd_pci_acp6x snd_hda_core snd_hwdep snd_pci_acp5x sp5100_tco snd_pcm k10temp snd_rn_pci_acp3x libarc4 i2c_piix4
5月 21 10:48:31 mi kernel:  snd_pci_acp3x snd_timer snd xhci_pci tpm_crb xhci_pci_renesas soundcore ccp cfg80211 ucsi_acpi typec_ucsi rfkill tpm_tis typec tpm_tis_core roles mac_hid i8042 video tpm i2c_hid_acpi wmi rng_core i2c_hid acpi_cpufreq pinctrl_amd amd_pmc pkcs8_key_parser dm_multipath dm_mod ledtrig_timer sg crypto_user fuse zram bpf_preload ip_tables x_tables amdgpu drm_ttm_helper ttm gpu_sched atkbd libps2 serio btrfs blake2b_generic libcrc32c crc32c_generic crc32c_intel xor raid6_pq
5月 21 10:48:31 mi kernel: ---[ end trace 0000000000000000 ]---
5月 21 10:48:31 mi kernel: RIP: 0010:nintendo_hid_event+0x5ff/0xd20 [hid_nintendo]
[... cut ...]
5月 21 10:49:02 mi kernel: RDX: 00007ffcd2231460 RSI: 00000000c0206440 RDI: 000000000000000e
5月 21 10:49:02 mi kernel: RBP: 00000000c0206440 R08: 0000000000000007 R09: 0000000000000005
5月 21 10:49:02 mi kernel: R10: 000055dec0cab010 R11: 0000000000000246 R12: 0000000000000002
5月 21 10:49:02 mi kernel: R13: 000000000000000e R14: 000000000011e000 R15: 0000000000000211
5月 21 10:49:02 mi kernel:  </TASK>
5月 21 10:49:30 mi kernel: watchdog: BUG: soft lockup - CPU#1 stuck for 49s! [Xorg:609]
```
</summary>

pickfire, 2022-05-21:
<details>
<summary>Mainline log `5.18.0-rc7-1-mainline`</summary>

```
...
5月 21 11:15:01 mi kernel: Bluetooth: hci0: AOSP extensions version v1.00
5月 21 11:15:01 mi kernel: Bluetooth: hci0: AOSP quality report is supported
5月 21 11:15:33 mi kernel: Bluetooth: HIDP (Human Interface Emulation) ver 1.2
5月 21 11:15:33 mi kernel: Bluetooth: HIDP socket layer initialized
5月 21 11:15:33 mi kernel: hid-generic 0005:057E:2009.0006: unknown main item tag 0x0
5月 21 11:15:33 mi kernel: input: Pro Controller as /devices/pci0000:00/0000:00:08.1/0000:02:00.3/usb1/1-4/1-4:1.0/bluetooth/hci0/hci0:1/0005:057E:2009.0006/input/input21
5月 21 11:15:33 mi kernel: hid-generic 0005:057E:2009.0006: input,hidraw5: BLUETOOTH HID v0.01 Gamepad [Pro Controller] on 80:30:49:10:6b:d0
5月 21 11:15:34 mi kernel: nintendo 0005:057E:2009.0006: unknown main item tag 0x0
5月 21 11:15:34 mi kernel: nintendo 0005:057E:2009.0006: hidraw5: BLUETOOTH HID v80.01 Gamepad [Pro Controller] on 80:30:49:10:6b:d0
5月 21 11:15:36 mi kernel: nintendo 0005:057E:2009.0006: using user cal for left stick
5月 21 11:15:36 mi kernel: nintendo 0005:057E:2009.0006: using user cal for right stick
5月 21 11:15:36 mi kernel: nintendo 0005:057E:2009.0006: using factory cal for IMU
5月 21 11:15:36 mi kernel: nintendo 0005:057E:2009.0006: controller MAC = 98:B6:E9:EC:11:08
5月 21 11:15:36 mi kernel: input: Nintendo Switch Pro Controller as /devices/pci0000:00/0000:00:08.1/0000:02:00.3/usb1/1-4/1-4:1.0/bluetooth/hci0/hci0:1/0005:057E:2009.0006/input/input22
5月 21 11:15:36 mi kernel: input: Nintendo Switch Pro Controller IMU as /devices/pci0000:00/0000:00:08.1/0000:02:00.3/usb1/1-4/1-4:1.0/bluetooth/hci0/hci0:1/0005:057E:2009.0006/input/input23
5月 21 11:15:36 mi kernel: divide error: 0000 [#1] PREEMPT SMP NOPTI
5月 21 11:15:36 mi kernel: CPU: 5 PID: 10745 Comm: khidpd_057e2009 Not tainted 5.18.0-rc7-1-mainline #1 65eeb77e9a84a6e1178324052e8f048ee7baa6f3
5月 21 11:15:36 mi kernel: Hardware name: TIMI RedmiBook 14 II/TM1951, BIOS RMARN4B0P0909 08/19/2020
5月 21 11:15:36 mi kernel: RIP: 0010:nintendo_hid_event+0x61c/0xd34 [hid_nintendo]
5月 21 11:15:36 mi kernel: Code: 08 00 00 8b bb 00 08 00 00 89 c6 0f b7 04 24 44 8b 83 04 08 00 00 39 c8 0f 8e 78 04 00 00 29 c8 29 cf 89 c2 c1 e0 0f 29 d0 99 <f7> ff 8b bb 0c 08 00 00 44 8b 8b 10 08 00 00 ba 01 80 ff ff 0f b7
5月 21 11:15:36 mi kernel: RSP: 0018:ffffa9b842befd48 EFLAGS: 00010212
5月 21 11:15:36 mi kernel: RAX: 000000000403f7f8 RBX: ffff981fabd91028 RCX: 0000000000000000
5月 21 11:15:36 mi kernel: RDX: 0000000000000000 RSI: 0000000000000807 RDI: 0000000000000000
5月 21 11:15:36 mi kernel: RBP: ffff981e883d0184 R08: 0000000000000000 R09: 0000000000000000
5月 21 11:15:36 mi kernel: R10: 0000000000000001 R11: 0000000000000005 R12: ffff981fabd91790
5月 21 11:15:36 mi kernel: R13: 0000000000000000 R14: ffff981e04362800 R15: ffff981fabd9176c
5月 21 11:15:36 mi kernel: FS:  0000000000000000(0000) GS:ffff9820ff740000(0000) knlGS:0000000000000000
5月 21 11:15:36 mi kernel: CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
5月 21 11:15:36 mi kernel: CR2: 000055725f3e7c40 CR3: 0000000296e10000 CR4: 0000000000350ee0
5月 21 11:15:36 mi kernel: Call Trace:
5月 21 11:15:36 mi kernel:  <TASK>
5月 21 11:15:36 mi kernel:  ? schedule+0x4b/0xb0
5月 21 11:15:36 mi kernel:  hid_input_report+0x139/0x160
5月 21 11:15:36 mi kernel:  hidp_session_run+0x357/0x6b0 [hidp 911721784933713739de472684805f11203fb951]
5月 21 11:15:36 mi kernel:  ? _raw_spin_rq_lock_irqsave+0x20/0x20
5月 21 11:15:36 mi kernel:  ? hidp_session_run+0x6b0/0x6b0 [hidp 911721784933713739de472684805f11203fb951]
5月 21 11:15:36 mi kernel:  hidp_session_thread+0x11d/0x230 [hidp 911721784933713739de472684805f11203fb951]
5月 21 11:15:36 mi kernel:  ? hidp_close+0x10/0x10 [hidp 911721784933713739de472684805f11203fb951]
5月 21 11:15:36 mi kernel:  ? hidp_close+0x10/0x10 [hidp 911721784933713739de472684805f11203fb951]
5月 21 11:15:36 mi kernel:  kthread+0xd7/0x100
5月 21 11:15:36 mi kernel:  ? kthread_complete_and_exit+0x20/0x20
5月 21 11:15:36 mi kernel:  ret_from_fork+0x1f/0x30
5月 21 11:15:36 mi kernel:  </TASK>
5月 21 11:15:36 mi kernel: Modules linked in: hid_nintendo ff_memless hidp snd_seq_dummy snd_hrtimer snd_seq snd_seq_device ccm rfcomm tun cmac algif_hash algif_skcipher af_alg bnep nls_iso8859_1 vfat fat hid_apple btusb btrtl btbcm btintel btmtk apple_mfi_fastcharge bluetooth usbhid ecdh_generic crc16 intel_rapl_msr snd_acp3x_pdm_dma joydev snd_soc_dmic snd_acp3x_rn mousedev intel_rapl_common serio_raw hid_multitouch edac_mce_amd wmi_bmof kvm_amd lzo_rle kvm irqbypass crct10dif_pclmul crc32_pclmul ghash_clmulni_intel aesni_intel crypto_simd cryptd rapl pcspkr rtw88_8822ce snd_sof_amd_renoir rtw88_8822c snd_sof_amd_acp snd_sof_pci snd_hda_codec_realtek snd_sof snd_hda_codec_generic snd_sof_utils rtw88_pci rtw88_core ledtrig_audio snd_soc_core snd_hda_codec_hdmi snd_compress snd_hda_intel mac80211 snd_intel_dspcfg ac97_bus snd_intel_sdw_acpi snd_pcm_dmaengine k10temp snd_hda_codec snd_pci_acp6x snd_pci_acp5x sp5100_tco snd_hda_core snd_rn_pci_acp3x snd_hwdep i2c_piix4 snd_acp_config snd_soc_acpi
5月 21 11:15:36 mi kernel:  libarc4 snd_pcm snd_pci_acp3x snd_timer snd xhci_pci ccp soundcore xhci_pci_renesas cfg80211 rfkill tpm_crb ucsi_acpi typec_ucsi tpm_tis tpm_tis_core typec roles i8042 mac_hid tpm i2c_hid_acpi video rng_core wmi i2c_hid acpi_cpufreq amd_pmc pinctrl_amd pkcs8_key_parser dm_multipath dm_mod ledtrig_timer sg crypto_user fuse zram bpf_preload ip_tables x_tables amdgpu drm_ttm_helper ttm gpu_sc [... cut ...]
5月 21 11:15:36 mi kernel: FS:  0000000000000000(0000) GS:ffff9820ff740000(0000) knlGS:0000000000000000
5月 21 11:15:36 mi kernel: CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
5月 21 11:15:36 mi kernel: CR2: 000055725f3e7c40 CR3: 0000000296e10000 CR4: 0000000000350ee0
5月 21 11:15:36 mi kernel: BUG: kernel NULL pointer dereference, address: 0000000000000000
5月 21 11:15:36 mi kernel: #PF: supervisor read access in kernel mode
5月 21 11:15:36 mi kernel: #PF: error_code(0x0000) - not-present page
```
</details>

dogtopus, 2022-05-21:
Looks like there's a division-by-zero error somewhere near `nintendo_hid_event`. Wild guess: something is wrong in the calibration data?

pickfire, 2022-05-21:
Not sure, I never tweaked it myself.

dogtopus, 2022-05-21:
> But I thought it was null pointer dereference? I can't see how is it division by zero.

Because

```
kernel: divide error: 0000 [#1] PREEMPT SMP NOPTI
```

pickfire, 2022-07-16:
I believe other people faced the same issue. https://bbs.archlinux.org/viewtopic.php?id=276196

jorgelukas, 2022-08-16:
still happen, same here with gulikit kingkong 2 pro, I wonder if it happens with the official switch pro 

Linux fedora 5.18.16-200.fc36.x86_64 #1 SMP PREEMPT_DYNAMIC Wed Aug 3 15:44:49 UTC 2022 x86_64 x86_64 x86_64 GNU/Linux

pickfire, 2022-08-17:
@jorgelukas you can try the method I suggested, blacklist `hid_nintendo` and use `hid-nintendo-dkms` instead, that one at least it won't hang, at least still usable.

JohnoKing, 2022-08-22:
I have debugged this hang and have found out that the `hid-nintendo-dkms` driver works because it contains the changes from this commit: https://github.com/nicman23/dkms-hid-nintendo/commit/7f984edddf7823055ecd5022e291d4a8c340ddbf. For some reason the kernel version of the `hid_nintendo` driver lacks those changes, which results in the hang when the GuliKit controller is in Nintendo Switch mode. Thus, while the hang can be avoided by using `hid-nintendo-dkms`, it's also possible to patch the kernel driver to avoid the hang. Patch against Linux 5.19.3:
```diff
# This patches the kernel driver with the changes in https://github.com/nicman23/dkms-hid-nintendo/commit/7f984edd
diff -Naur linux-5.19.3/drivers/hid/hid-nintendo.c linux-5.19.3-fix/drivers/hid/hid-nintendo.c
--- linux-5.19.3/drivers/hid/hid-nintendo.c	2022-08-21 06:20:09.000000000 -0700
+++ linux-5.19.3-fix/drivers/hid/hid-nintendo.c	2022-08-22 14:44:11.876342940 -0700
@@ -794,7 +794,14 @@
 					    &ctlr->left_stick_cal_x,
 					    &ctlr->left_stick_cal_y,
 					    true);
-	if (ret) {
+
+	/* check whether read succeeded and perform plausibility check for retrieved values */
+	if (ret ||
+		ctlr->left_stick_cal_x.min >= ctlr->left_stick_cal_x.center ||
+		ctlr->left_stick_cal_x.center >= ctlr->left_stick_cal_x.max ||
+		ctlr->left_stick_cal_y.min >= ctlr->left_stick_cal_y.center ||
+		ctlr->left_stick_cal_y.center >= ctlr->left_stick_cal_y.max
+	) {
 		hid_warn(ctlr->hdev,
 			 "Failed to read left stick cal, using dflts; e=%d\n",
 			 ret);
@@ -813,7 +820,14 @@
 					    &ctlr->right_stick_cal_x,
 					    &ctlr->right_stick_cal_y,
 					    false);
-	if (ret) {
+
+	/* check whether read succeeded and perform plausibility check for retrieved values */
+	if (ret ||
+		ctlr->right_stick_cal_x.min >= ctlr->right_stick_cal_x.center ||
+		ctlr->right_stick_cal_x.center >= ctlr->right_stick_cal_x.max ||
+		ctlr->right_stick_cal_y.min >= ctlr->right_stick_cal_y.center ||
+		ctlr->right_stick_cal_y.center >= ctlr->right_stick_cal_y.max
+	) {
 		hid_warn(ctlr->hdev,
 			 "Failed to read right stick cal, using dflts; e=%d\n",
 			 ret);
```
Recompiling the kernel with the above patch allows the GuliKit controller to work without installing `hid-nintendo-dkms`. Granted, I'm not sure if this is the best way to fix the hang.

pickfire, 2022-08-25:
@JohnoKing Interesting, if you are free, do you mind sharing you did you debug it and test it out? Did you recompile the linux kernel with that patch?

If it works, should I go and send a patch to linux kernel? I am not familiar with that.

JohnoKing, 2022-08-25:
To debug the hang, I created a diff of the [kernel](https://github.com/torvalds/linux/blob/master/drivers/hid/hid-nintendo.c) and [dkms](https://github.com/nicman23/dkms-hid-nintendo/blob/master/src/hid-nintendo.c) versions of hid-nintendo.c with the following command:
```
diff -u hid-nintendo.c-kernel hid-nintendo.c-dkms
```
After that, I tried applying the various individual changes in the kernel version to the dkms version of hid-nintendo, seeing what worked and what would cause a crash. These changes would be tested by recompiling the dkms driver, unloading it with `modprobe -r hid-nintendo` and reloading it with `modprobe hid-nintendo`. I used this script to aid in doing that (normally it's for using my Pro Controller with Steam, though in this case it's useful for debugging):

<details>
<summary>Script (dropdown):</summary>

```bash
# Hopefully this becomes unnecessary when Steam updates to support kernel 5.16+.
# Until then Steam itself cannot use hid-nintendo.

if [[ $(id -u) != 0 ]]; then
	echo "This script must be run as root!"
	exit 1
fi

if [[ -w /etc/modprobe.d/blacklist.conf.disablesteam ]] && [[ -w /usr/lib/udev/rules.d/89-joycond.rules.disablesteam ]]; then
	# Enable hid-nintendo and joycond-cemuhook
	mv /etc/modprobe.d/blacklist.conf /etc/modprobe.d/blacklist.conf.enablesteam
	mv /etc/modprobe.d/blacklist.conf.disablesteam /etc/modprobe.d/blacklist.conf
	modprobe hid-nintendo
	mv /usr/lib/udev/rules.d/89-joycond.rules /usr/lib/udev/rules.d/89-joycond.rules.enablesteam
	mv /usr/lib/udev/rules.d/89-joycond.rules.disablesteam /usr/lib/udev/rules.d/89-joycond.rules
	udevadm control --reload-rules
	udevadm trigger
	echo "Enabled hid-nintendo and joycond-cemuhook support"
elif [[ -w /etc/modprobe.d/blacklist.conf.enablesteam ]] && [[ -w /usr/lib/udev/rules.d/89-joycond.rules.enablesteam ]]; then
	# Enable Steam support
	mv /etc/modprobe.d/blacklist.conf /etc/modprobe.d/blacklist.conf.disablesteam
	mv /etc/modprobe.d/blacklist.conf.enablesteam /etc/modprobe.d/blacklist.conf
	modprobe -r hid-nintendo
	mv /usr/lib/udev/rules.d/89-joycond.rules /usr/lib/udev/rules.d/89-joycond.rules.disablesteam
	mv /usr/lib/udev/rules.d/89-joycond.rules.enablesteam /usr/lib/udev/rules.d/89-joycond.rules
	udevadm control --reload-rules
	udevadm trigger
	echo "Enabled Steam support"
else
	# Should not happen
	echo "INTERNAL ERROR"
	exit 128
fi

# Later addition: For debugging, sync the filesystem in case the compiled kernel module hangs
sync
```

</details>

After I determined which change in the kernel version caused the hang, I uninstalled hid-nintendo-dkms and applied the resulting [patch](https://github.com/DanielOgorchock/linux/issues/36#issuecomment-1223165318) to the Linux kernel. Then I compiled the patched kernel and installed it, then rebooted the system. I plugged in the GuliKit controller in Nintendo Switch mode and got no hang with the patched kernel driver (I tested the controller in yuzu with the patched kernel and it worked just fine).

> If it works, should I go and send a patch to linux kernel? I am not familiar with that.

I'm not familiar with sending patches to the Linux kernel either. In the case of the patch (which is from https://github.com/nicman23/dkms-hid-nintendo/pull/25), it looks like it was supposed to have been sent [over a year ago](https://github.com/DanielOgorchock/linux/issues/21), but it still hasn't been submitted :/.

JohnoKing, 2022-10-08:
The bugfix for the GuliKit controller hang is now set to be included in Linux 6.1. See:
https://github.com/torvalds/linux/commit/79d11de9637a37035b4a72ccb528b3dfebc1bf67
https://github.com/torvalds/linux/commit/50503e360eeb968a3d00234c9cc4057d774c3e9a