DanielOgorchock/linux #39: Is hid-nintendo still maintained?

SuperSamus, 2022-05-30:
`hid-nintendo` has some major problems (made worse by the fact that it's mainlined), but despite that, the maintainer doesn't have any form of activity on GitHub since September 2021.
It's perfectly fine if they don't have any spare time to work on it, but the status is still `Maintained` on the [maintainer list](https://github.com/torvalds/linux/blob/b00ed48bb0a7c295facf9036135a573a5cdbe7de/MAINTAINERS#L14009), and it should be updated if it's not actually maintained anymore.

DanielOgorchock, 2022-05-30:
Hey sorry for the lack of responsiveness on GitHub. Now that the driver's been mainlined, I haven't been checking in on GitHub much. 

I think we should encourage having most correspondence occur on the linux-input mailing list now, including patches/features.

SuperSamus, 2022-05-30:
Ok. Sorry for assuming so.
Then I think there should be something like a pinned issue (this one?) or a README for this.

Also, some of the issues here should also be reported on [Bugzilla](https://bugzilla.kernel.org/).

SuperSamus, 2022-07-10:
> So then, where else can we submit our issues and be attended more efficiently, if not here? The Bugzilla?

I guess so?
[Searching "nintendo" on Bugzilla](https://bugzilla.kernel.org/buglist.cgi?bug_status=NEW&bug_status=ASSIGNED&bug_status=REOPENED&bug_status=RESOLVED&bug_status=VERIFIED&bug_status=REJECTED&bug_status=DEFERRED&bug_status=NEEDINFO&bug_status=CLOSED&list_id=1116807&query_format=advanced&short_desc=nintendo&short_desc_type=allwordssubstr) returns 5 results, 2 of which being open despite being obsolete, and 3 of them mine (ports of issues of this repo).