DanielOgorchock/linux #41: Player LED behavior is wrong

SuperSamus, 2022-07-09:
(Copied from https://bugzilla.kernel.org/show_bug.cgi?id=216225)

The behavior of the Nintendo Switch Pro Controller player LED on Linux is this: the first time it's connected the first led turns on, the second time it's connected the first two LEDs (not just the second) turn on, the third time the first three LEDs (not just the third) turn on, the fourth time all LEDs (not just the fourth) turn on, and then it cycles back to the beginning.

This is completely different from the behavior when connected to the Switch console, where only one player LED ever turns on, and it's the one associated with the player (first LED for player one, second LED for player two...), not the number of times it's been connected.

Joycond does fix this, but there shouldn't be the need of using a daemon for something simple like this.

I don't know if from the kernel it's possible to get the "player number", but it would still be an improvement to only have one LED turning on.

Tested on a Pro Controller 050000007e0500000920000001800000, don't own a Joy-Con.

Finally, this is just my opinion.

SuperSamus, 2022-07-10:
I gave a look to [hid-nintendo.c](https://github.com/DanielOgorchock/linux/blob/ogorchock/drivers/hid/hid-nintendo.c) to see why it behaves like this.
The static variable that controls the number is `input_num`.
It increases by 1 (and resets to 1 when it reaches 5) every time after LEDs are initialized (`joycon_leds_create`). It never decreases.
Finally, the reason why multiple LEDs are turned on instead of just one, seems to be in `led->brightness = ((i + 1) <= input_num) ? 1 : 0;` (it's a `<=` instead of a `==`).

Instead of doing this, would it be possible to have a static variable that increases by 1 when a controller is connected (`nintendo_hid_probe`) and decreases by 1 when disconnected `nintendo_hid_remove`? (With the necessary mutexes, of course.)

I don't know how to test kernel stuff, so that's why I'm just leaving this comment.

SuperSamus, 2022-07-11:
I just realized that using one global counter would have a problem.
If two controllers are connected, player one disconnects, then a controller is connected, they will both be "player two" according to the leds.

The solution would be to store the controllers in a list, and when a controller disconnects either:
 - All the player of a higher number will shift to the left, and the leds updated accordingly;
 - Leave an empty slot. When a controller is connected, it takes the first empty slot available (this is what's done on the Switch console).

JohnoKing, 2022-09-25:
I have begun working on a patch to implement this LED number functionality. The initial version of the patch is available below for testing (the code will need some cleanup before being submitted to the LKML):
```diff
diff --git a/drivers/hid/hid-nintendo.c b/drivers/hid/hid-nintendo.c
index 5bfc0c450460..3141f89d2a21 100644
--- a/drivers/hid/hid-nintendo.c
+++ b/drivers/hid/hid-nintendo.c
@@ -1836,6 +1836,8 @@ static int joycon_home_led_brightness_set(struct led_classdev *led,
 }
 
 static DEFINE_MUTEX(joycon_input_num_mutex);
+static bool input_num[16];
+
 static int joycon_leds_create(struct joycon_ctlr *ctlr)
 {
 	struct hid_device *hdev = ctlr->hdev;
@@ -1845,12 +1847,30 @@ static int joycon_leds_create(struct joycon_ctlr *ctlr)
 	char *name;
 	int ret = 0;
 	int i;
-	static int input_num = 1;
+	int current_input_num;
 
 	/* Set the default controller player leds based on controller number */
 	mutex_lock(&joycon_input_num_mutex);
 	mutex_lock(&ctlr->output_mutex);
-	ret = joycon_set_player_leds(ctlr, 0, 0xF >> (4 - input_num));
+
+	/* use first available LED */
+	for (i = 0; i < 17; i++) {
+		if (i == 17) {
+			ret = -EOVERFLOW;
+			hid_warn(ctlr->hdev, "Failed to set leds; ret=%d\n",
+				 ret);
+		}
+		if (!input_num[i]) {
+			input_num[i] = true;
+			if (i < 4)
+				current_input_num = i + 1;
+			else
+				current_input_num = (i + 1) % 4 ? (i + 1) % 4 : 4;
+			break;
+		}
+	}
+
+	ret = joycon_set_player_leds(ctlr, 0, 0xF >> (4 - current_input_num));
 	if (ret)
 		hid_warn(ctlr->hdev, "Failed to set leds; ret=%d\n", ret);
 	mutex_unlock(&ctlr->output_mutex);
@@ -1868,7 +1888,7 @@ static int joycon_leds_create(struct joycon_ctlr *ctlr)
 
 		led = &ctlr->leds[i];
 		led->name = name;
-		led->brightness = ((i + 1) <= input_num) ? 1 : 0;
+		led->brightness = ((i + 1) <= current_input_num) ? 1 : 0;
 		led->max_brightness = 1;
 		led->brightness_set_blocking =
 					joycon_player_led_brightness_set;
@@ -1881,9 +1901,6 @@ static int joycon_leds_create(struct joycon_ctlr *ctlr)
 			return ret;
 		}
 	}
-
-	if (++input_num > 4)
-		input_num = 1;
 	mutex_unlock(&joycon_input_num_mutex);
 
 	/* configure the home LED */
@@ -2292,6 +2309,7 @@ static void nintendo_hid_remove(struct hid_device *hdev)
 {
 	struct joycon_ctlr *ctlr = hid_get_drvdata(hdev);
 	unsigned long flags;
+	int i;
 
 	hid_dbg(hdev, "remove\n");
 
@@ -2302,6 +2320,13 @@ static void nintendo_hid_remove(struct hid_device *hdev)
 
 	destroy_workqueue(ctlr->rumble_queue);
 
+	for (i = 0; i < 16; i++) {
+		if (input_num[i]) {
+			input_num[i] = false;
+			break;
+		}
+	}
+
 	hid_hw_close(hdev);
 	hid_hw_stop(hdev);
 }

```

SuperSamus, 2024-07-22:
Fixed in Linux 6.10.

closed 2024-07-22 by SuperSamus