DanielOgorchock/linux #42: Pro Controller connected with USB, doesn't reconnect after computer resumes from suspend

SuperSamus, 2022-07-10:
Normally, if you connect a Nintendo Switch Pro Controller via USB, it is automatically connected when turning the computer on.
If however the computer is suspended, when resuming the computer, the controller won't reconnect even if you press a button. The only way to reconnect is either to press the sync button on the controller (which normally powers it off), unplug and plug the cable, or reboot the computer.

Step by step to reproduce:

- Connect the controller via USB (the controller will be connected);
- Suspend the computer;
- Resume the computer;
- The controller won't be connected (even if you press buttons, only the sync button will make it work again).

DanielOgorchock, 2022-11-04:
Thanks for pointing this out. Yeah the driver doesn't have pm hooks implemented. I should really add those.

SuperSamus, 2023-04-21:
Hi, is there any update on this?

The template for should be mostly like this, right? (Shamelessly copied from [hid-sony](https://github.com/torvalds/linux/blob/master/drivers/hid/hid-sony.c#L2182-L2217))

<details>
<summary>Template</summary>

```c
#ifdef CONFIG_PM

static int nintendo_hid_suspend(struct hid_device *hdev, pm_message_t message)
{
#ifdef CONFIG_NINTENDO_FF

	/* On suspend stop any running force-feedback events (I don't know if it is necessary on Nintendo) */
	struct joycon_ctlr *ctlr = hid_get_drvdata(hdev);
        /* etc */

#endif
	return 0;
}

static int nintendo_hid_resume(struct hid_device *hdev)
{
	/*
	 * The controllers on USB need to be reinitialized
	 * on resume or they won't behave properly.
	 */
	if (hdev->bus == BUS_USB) {
		/* Instructions to resume */
	}

	return 0;
}

#endif

/* ... */

static struct hid_driver nintendo_hid_driver = {
	.name		= "nintendo",
	.id_table	= nintendo_hid_devices,
	.probe		= nintendo_hid_probe,
	.remove		= nintendo_hid_remove,
	.raw_event	= nintendo_hid_event,

#ifdef CONFIG_PM
	.suspend          = nintendo_hid_suspend,
	.resume	          = nintendo_hid_resume,
	.reset_resume     = nintendo_hid_resume,
#endif
};
```
</details>

SuperSamus, 2023-07-09:
Well, I tried it myself.
After trying to compile the Linux kernel to load a module (first time), I finally did it.
I didn't test it much, but it seems to work well.
Here is the patch: [pro-controller-suspend.txt](https://github.com/DanielOgorchock/linux/files/11996278/pro-controller-suspend.txt)

If no one has anything against it, I'll send it after a week. (My first kernel contribution)

SuperSamus, 2023-07-20:
https://lore.kernel.org/linux-input/20230712121042.25318-2-tinozzo123@gmail.com/
No one has reviewed it yet.

SuperSamus, 2023-12-10:
Fixed on Linux 6.6.

closed 2023-12-10 by SuperSamus