DanielOgorchock/linux #45: Bluetooth is refusing to work

michaeljhkim, 2023-04-13:
So the USB system works 100% fine, but trying to get my pro controller to work with bluetooth is impossible. 

It says that it has connected through bluetooth, but it is not recognized as an input device of any kind. My controller works if I open up steam, but steam uses it's own input driver system of course.

I can confidentially say that hid_nintendo has a problem with bluetooth connection.

michaeljhkim, 2023-04-13:
here is the dmesg:

[96187.539804] nintendo 0003:057E:2009.0031: hidraw8: USB HID v81.11 Joystick [Nintendo Co., Ltd. Pro Controller] on usb-0000:00:14.0-5/input0
[96187.860646] nintendo 0003:057E:2009.0031: using user cal for left stick
[96187.876645] nintendo 0003:057E:2009.0031: using user cal for right stick
[96187.892648] nintendo 0003:057E:2009.0031: Failed to read left stick cal, using defaults; e=-22
[96187.908645] nintendo 0003:057E:2009.0031: Failed to read right stick cal, using defaults; e=-22
[96187.924639] nintendo 0003:057E:2009.0031: using factory cal for IMU
[96188.004651] nintendo 0003:057E:2009.0031: controller MAC = 98:B6:E9:57:EB:0F
[96188.036714] input: Nintendo Switch Pro Controller as /devices/pci0000:00/0000:00:14.0/usb1/1-5/1-5:1.0/0003:057E:2009.0031/input/input82
[96188.036931] input: Nintendo Switch Pro Controller IMU as /devices/pci0000:00/0000:00:14.0/usb1/1-5/1-5:1.0/0003:057E:2009.0031/input/input83
[96299.561086] usb 1-5: USB disconnect, device number 8
[100925.038131] input: Aeropex by AfterShokz (AVRCP) as /devices/virtual/input/input84
[100964.254031] usb 1-5: new high-speed USB device number 9 using xhci_hcd
[100964.402236] usb 1-5: New USB device found, idVendor=04e8, idProduct=6860, bcdDevice= 5.04
[100964.402239] usb 1-5: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[100964.402240] usb 1-5: Product: SAMSUNG_Android
[100964.402241] usb 1-5: Manufacturer: SAMSUNG
[100964.402242] usb 1-5: SerialNumber: R5CT71M0P0A
[100964.425989] cdc_acm 1-5:1.1: ttyACM0: USB ACM device
[100964.426016] usbcore: registered new interface driver cdc_acm
[100964.426023] cdc_acm: USB Abstract Control Model driver for USB modems and ISDN adapters
[100971.097749] nintendo 0005:057E:2009.0032: unknown main item tag 0x0
[100971.098014] nintendo 0005:057E:2009.0032: hidraw8: BLUETOOTH HID v80.01 Gamepad [Pro Controller] on d4:54:8b:d8:74:2f
[100975.153917] nintendo 0005:057E:2009.0032: failed reading SPI flash; ret=-110
[100975.153936] nintendo 0005:057E:2009.0032: using factory cal for left stick
[100977.180491] nintendo 0005:057E:2009.0032: failed reading SPI flash; ret=-110
[100977.180498] nintendo 0005:057E:2009.0032: using factory cal for right stick
[100978.156364] usb 1-5: USB disconnect, device number 9
[100979.207158] nintendo 0005:057E:2009.0032: failed reading SPI flash; ret=-110
[100979.207164] nintendo 0005:057E:2009.0032: Failed to read left stick cal, using defaults; e=-110
[100981.233799] nintendo 0005:057E:2009.0032: failed reading SPI flash; ret=-110
[100981.233805] nintendo 0005:057E:2009.0032: Failed to read right stick cal, using defaults; e=-110
[100983.260497] nintendo 0005:057E:2009.0032: failed reading SPI flash; ret=-110
[100983.260516] nintendo 0005:057E:2009.0032: using factory cal for IMU
[100985.287103] nintendo 0005:057E:2009.0032: failed reading SPI flash; ret=-110
[100985.287122] nintendo 0005:057E:2009.0032: Failed to read IMU cal, using defaults; ret=-110
[100985.287128] nintendo 0005:057E:2009.0032: Unable to read IMU calibration data
[100987.313812] nintendo 0005:057E:2009.0032: Failed to set report mode; ret=-110
[100987.314165] nintendo 0005:057E:2009.0032: probe - fail = -110
[100987.314179] nintendo: probe of 0005:057E:2009.0032 failed with error -110
[101735.055818] nintendo 0005:057E:2009.0033: unknown main item tag 0x0
[101735.056245] nintendo 0005:057E:2009.0033: hidraw8: BLUETOOTH HID v80.01 Gamepad [Pro Controller] on d4:54:8b:d8:74:2f
[101739.114971] nintendo 0005:057E:2009.0033: failed reading SPI flash; ret=-110
[101739.114988] nintendo 0005:057E:2009.0033: using factory cal for left stick
[101741.144927] nintendo 0005:057E:2009.0033: failed reading SPI flash; ret=-110
[101741.144944] nintendo 0005:057E:2009.0033: using factory cal for right stick
[101743.168176] nintendo 0005:057E:2009.0033: failed reading SPI flash; ret=-110
[101743.168188] nintendo 0005:057E:2009.0033: Failed to read left stick cal, using defaults; e=-110
[101745.194859] nintendo 0005:057E:2009.0033: failed reading SPI flash; ret=-110
[101745.194876] nintendo 0005:057E:2009.0033: Failed to read right stick cal, using defaults; e=-110
[101747.221501] nintendo 0005:057E:2009.0033: failed reading SPI flash; ret=-110
[101747.221517] nintendo 0005:057E:2009.0033: using factory cal for IMU
[101749.251387] nintendo 0005:057E:2009.0033: failed reading SPI flash; ret=-110
[101749.251403] nintendo 0005:057E:2009.0033: Failed to read IMU cal, using defaults; ret=-110
[101749.251407] nintendo 0005:057E:2009.0033: Unable to read IMU calibration data
[101751.274771] nintendo 0005:057E:2009.0033: Failed to set report mode; ret=-110
[101751.274984] nintendo 0005:057E:2009.0033: probe - fail = -110
[101751.274992] nintendo: probe of 0005:057E:2009.0033 failed with error -110
[103893.019787] input: Aeropex by AfterShokz (AVRCP) as /devices/virtual/input/input85
[104198.914932] nintendo 0005:057E:2009.0034: unknown main item tag 0x0
[104198.915231] nintendo 0005:057E:2009.0034: hidraw8: BLUETOOTH HID v80.01 Gamepad [Pro Controller] on d4:54:8b:d8:74:2f
[104202.967894] nintendo 0005:057E:2009.0034: failed reading SPI flash; ret=-110
[104202.967910] nintendo 0005:057E:2009.0034: using factory cal for left stick
[104204.994530] nintendo 0005:057E:2009.0034: failed reading SPI flash; ret=-110
[104204.994547] nintendo 0005:057E:2009.0034: using factory cal for right stick
[104207.021162] nintendo 0005:057E:2009.0034: failed reading SPI flash; ret=-110
[... cut ...]
[108350.100329] nintendo 0005:057E:2009.003B: using factory cal for IMU
[108352.126937] nintendo 0005:057E:2009.003B: failed reading SPI flash; ret=-110
[108352.126958] nintendo 0005:057E:2009.003B: Failed to read IMU cal, using defaults; ret=-110
[108352.126964] nintendo 0005:057E:2009.003B: Unable to read IMU calibration data
[108354.153560] nintendo 0005:057E:2009.003B: Failed to set report mode; ret=-110
[108354.153818] nintendo 0005:057E:2009.003B: probe - fail = -110
[108354.153826] nintendo: probe of 0005:057E:2009.003B failed with error -110

EuanFH, 2023-05-24:
yeah I'm getting the exact same issue. looking at the driver it seems to be timing out waiting for a sub command to run.

osalas88, 2023-08-12:
I'm also using an non-official nintendo controller (8bitdo Pro 2 Controller)
```
[Sat Aug 12 02:54:17 2023] nintendo 0005:057E:2009.0009: unknown main item tag 0x0
[Sat Aug 12 02:54:17 2023] nintendo 0005:057E:2009.0009: hidraw4: BLUETOOTH HID v80.01 Gamepad [Pro Controller] on 44:af:28:13:a0:91
[Sat Aug 12 02:54:21 2023] nintendo 0005:057E:2009.0009: failed reading SPI flash; ret=-110
[Sat Aug 12 02:54:21 2023] nintendo 0005:057E:2009.0009: using factory cal for left stick
[Sat Aug 12 02:54:23 2023] nintendo 0005:057E:2009.0009: failed reading SPI flash; ret=-110
[Sat Aug 12 02:54:23 2023] nintendo 0005:057E:2009.0009: using factory cal for right stick
[Sat Aug 12 02:54:25 2023] nintendo 0005:057E:2009.0009: failed reading SPI flash; ret=-110
[Sat Aug 12 02:54:25 2023] nintendo 0005:057E:2009.0009: Failed to read left stick cal, using defaults; e=-110
[Sat Aug 12 02:54:27 2023] nintendo 0005:057E:2009.0009: failed reading SPI flash; ret=-110
[Sat Aug 12 02:54:27 2023] nintendo 0005:057E:2009.0009: Failed to read right stick cal, using defaults; e=-110
[Sat Aug 12 02:54:29 2023] nintendo 0005:057E:2009.0009: failed reading SPI flash; ret=-110
[Sat Aug 12 02:54:29 2023] nintendo 0005:057E:2009.0009: using factory cal for IMU
[Sat Aug 12 02:54:31 2023] nintendo 0005:057E:2009.0009: failed reading SPI flash; ret=-110
[Sat Aug 12 02:54:31 2023] nintendo 0005:057E:2009.0009: Failed to read IMU cal, using defaults; ret=-110
[Sat Aug 12 02:54:31 2023] nintendo 0005:057E:2009.0009: Unable to read IMU calibration data
[Sat Aug 12 02:54:33 2023] nintendo 0005:057E:2009.0009: Failed to set report mode; ret=-110
[Sat Aug 12 02:54:33 2023] nintendo 0005:057E:2009.0009: probe - fail = -110
[Sat Aug 12 02:54:33 2023] nintendo: probe of 0005:057E:2009.0009 failed with error -110
```

osalas88, 2023-08-12:
So it seems our problems are a duplicate of another issue here, there is a workaround:

https://github.com/DanielOgorchock/linux/issues/10#issuecomment-1321074341

but I have not tried it myself.

EuanFH, 2023-08-12:
No this is a separate issue I'm pretty sure the vendor id and product id is correct for the king long 2 controller it copies Nintendo's official ones.

osalas88, 2023-08-12:
Could you post then your dmesg?