DanielOgorchock/linux #51: 3rd Party Pro Controller SP5248LED

juampe, 2024-02-22:
3rd Party Pro Controller SP5248LED (NYYXI CHAOS) with a N100 Mini PC

Compiled with https://github.com/nicman23/dkms-hid-nintendo (wait please) and this repo (hid source filesfiles related) in order to simplify testing

```
root@nuci:/usr/src# git clone https://github.com/nicman23/dkms-hid-nintendo
Clonando en 'dkms-hid-nintendo'...
remote: Enumerating objects: 410, done.
remote: Counting objects: 100% (48/48), done.
remote: Compressing objects: 100% (28/28), done.
remote: Total 410 (delta 30), reused 23 (delta 20), pack-reused 362
Recibiendo objetos: 100% (410/410), 688.65 KiB | 3.02 MiB/s, listo.
Resolviendo deltas: 100% (72/72), listo.
root@nuci:/usr/src# cd dkms-hid-nintendo/
root@nuci:/usr/src#for i in hid-ids.h  hid-nintendo.c  hidraw.c  uhid.c
do
        wget https://raw.githubusercontent.com/DanielOgorchock/linux/ogorchock/drivers/hid/$i -O src/$i
done
root@nuci:/usr/src/dkms-hid-nintendo# dkms build nintendo -v 3.0 --force                                                                                        Deprecated feature: REMAKE_INITRD (/var/lib/dkms/nintendo/3.0/source/dkms.conf)
Deprecated feature: REMAKE_INITRD (/etc/dkms/framework.conf)
Sign command: /usr/bin/kmodsign
Signing key: /var/lib/shim-signed/mok/MOK.priv
Public certificate (MOK): /var/lib/shim-signed/mok/MOK.der

Building module:
Cleaning build area...
make -j4 KERNELRELEASE=6.5.0-17-generic -C /lib/modules/6.5.0-17-generic/build M=/var/lib/dkms/nintendo/3.0/build/src modules...
Signing module /var/lib/dkms/nintendo/3.0/build/src/hid-nintendo.ko
Cleaning build area...
root@nuci:/usr/src/dkms-hid-nintendo# rmmod hid_nintendo
root@nuci:/usr/src/dkms-hid-nintendo# modprobe -a /lib/modules/6.5.0-17-generic/updates/dkms/hid-nintendo.ko.zst
```
Power up the controller
```
feb 22 11:53:31 nuci kernel: nintendo 0005:057E:2009.0006: unknown main item tag 0x0
feb 22 11:53:31 nuci kernel: nintendo 0005:057E:2009.0006: hidraw0: BLUETOOTH HID v80.01 Gamepad [Pro Controller] on c8:8a:d8:11:d0:51
feb 22 11:53:35 nuci kernel: nintendo 0005:057E:2009.0006: failed reading SPI flash; ret=-110
feb 22 11:53:35 nuci kernel: nintendo 0005:057E:2009.0006: using factory cal for left stick
feb 22 11:53:37 nuci kernel: nintendo 0005:057E:2009.0006: failed reading SPI flash; ret=-110
feb 22 11:53:37 nuci kernel: nintendo 0005:057E:2009.0006: using factory cal for right stick
feb 22 11:53:39 nuci kernel: nintendo 0005:057E:2009.0006: failed reading SPI flash; ret=-110
feb 22 11:53:39 nuci kernel: nintendo 0005:057E:2009.0006: Failed to read left stick cal, using dflts; e=-110
feb 22 11:53:41 nuci kernel: nintendo 0005:057E:2009.0006: failed reading SPI flash; ret=-110
feb 22 11:53:41 nuci kernel: nintendo 0005:057E:2009.0006: Failed to read right stick cal, using dflts; e=-110
feb 22 11:53:43 nuci kernel: nintendo 0005:057E:2009.0006: failed reading SPI flash; ret=-110
feb 22 11:53:43 nuci kernel: nintendo 0005:057E:2009.0006: using factory cal for IMU
feb 22 11:53:45 nuci kernel: nintendo 0005:057E:2009.0006: failed reading SPI flash; ret=-110
feb 22 11:53:45 nuci kernel: nintendo 0005:057E:2009.0006: Failed to read IMU cal, using defaults; ret=-110
feb 22 11:53:45 nuci kernel: nintendo 0005:057E:2009.0006: Unable to read IMU calibration data
feb 22 11:53:45 nuci kernel: ================================================================================
feb 22 11:53:45 nuci kernel: UBSAN: array-index-out-of-bounds in /var/lib/dkms/nintendo/3.0/build/src/hid-nintendo.c:905:11
feb 22 11:53:45 nuci kernel: index 0 is out of range for type 'u8 [*]'
feb 22 11:53:45 nuci kernel: CPU: 0 PID: 1709558 Comm: kworker/0:2 Tainted: G           OE      6.5.0-17-generic #17-Ubuntu
feb 22 11:53:45 nuci kernel: Hardware name:  /, BIOS 5.26 09/26/2023
feb 22 11:53:45 nuci kernel: Workqueue: events uhid_device_add_worker [uhid]
feb 22 11:53:45 nuci kernel: Call Trace:
feb 22 11:53:45 nuci kernel:  <TASK>
feb 22 11:53:45 nuci kernel:  dump_stack_lvl+0x48/0x70
feb 22 11:53:45 nuci kernel:  dump_stack+0x10/0x20
feb 22 11:53:45 nuci kernel:  __ubsan_handle_out_of_bounds+0xc6/0x110
feb 22 11:53:45 nuci kernel:  nintendo_hid_probe+0x432/0x1a80 [hid_nintendo]
feb 22 11:53:45 nuci kernel:  ? devres_open_group+0x53/0x1b0
feb 22 11:53:45 nuci kernel:  hid_device_probe+0x12d/0x1b0 [hid]
feb 22 11:53:45 nuci kernel:  really_probe+0x1c4/0x410
feb 22 11:53:45 nuci kernel:  __driver_probe_device+0x8c/0x180
feb 22 11:53:45 nuci kernel:  driver_probe_device+0x24/0xd0
feb 22 11:53:45 nuci kernel:  __device_attach_driver+0xcd/0x170
feb 22 11:53:45 nuci kernel:  ? __pfx___device_attach_driver+0x10/0x10
feb 22 11:53:45 nuci kernel:  bus_for_each_drv+0x94/0xf0
feb 22 11:53:45 nuci kernel:  __device_attach+0xb6/0x1d0
feb 22 11:53:45 nuci kernel:  device_initial_probe+0x13/0x20
feb 22 11:53:45 nuci kernel:  bus_probe_device+0x9f/0xb0
feb 22 11:53:45 nuci kernel:  device_add+0x50b/0x700
feb 22 11:53:45 nuci kernel:  hid_add_device+0xf9/0x2d0 [hid]
feb 22 11:53:45 nuci kernel:  uhid_device_add_worker+0x19/0x80 [uhid]
feb 22 11:53:45 nuci kernel:  process_one_work+0x220/0x440
feb 22 11:53:45 nuci kernel:  worker_thread+0x4d/0x3f0
feb 22 11:53:45 nuci kernel:  ? _raw_spin_lock_irqsave+0xe/0x20
feb 22 11:53:45 nuci kernel:  ? __pfx_worker_thread+0x10/0x10
feb 22 11:53:45 nuci kernel:  kthread+0xef/0x120
feb 22 11:53:45 nuci kernel:  ? __pfx_kthread+0x10/0x10
feb 22 11:53:45 nuci kernel:  ret_from_fork+0x44/0x70
feb 22 11:53:45 nuci kernel:  ? __pfx_kthread+0x10/0x10
feb 22 11:53:45 nuci kernel:  ret_from_fork_asm+0x1b/0x30
feb 22 11:53:45 nuci kernel:  </TASK>
feb 22 11:53:45 nuci kernel: ================================================================================
feb 22 11:53:47 nuci kernel: nintendo 0005:057E:2009.0006: Failed to set report mode; ret=-110
feb 22 11:53:47 nuci kernel: nintendo 0005:057E:2009.0006: probe - fail = -110
feb 22 11:53:47 nuci kernel: nintendo: probe of 0005:057E:2009.0006 failed with error -110
```

juampe, 2024-02-22:
Add diferencial bluetooth network analisis via wireshark files of a good operation (windows) and a bad operation (linux)
[image]
[nyxi-good-vs-bad.zip](https://github.com/DanielOgorchock/linux/files/14375622/nyxi-good-vs-bad.zip)

moving-hawk, 2024-03-30:
I also use a 3rd party gamepad and My Linux cannot detect it correctly. 
```
[   40.051886] nintendo 0003:057E:2009.0002: hidraw1: USB HID v81.11 Joystick [BEITONG  BEITONG A1N3 NS GAMEPAD ] on usb-0000:04:00.4-2/input0
[   42.150813] nintendo 0003:057E:2009.0002: Failed to set baudrate; ret=-110
[   42.150820] nintendo 0003:057E:2009.0002: Failed to initialize controller; ret=-110
[   42.151067] nintendo 0003:057E:2009.0002: probe - fail = -110
[   42.151079] nintendo: probe of 0003:057E:2009.0002 failed with error -110
```