DanielOgorchock/linux #54: dkms-hid-nintendo: Nintendo Switch Pro Controller stops working after Suspend on MATE Desktop

davidhedlund, 2024-10-02:
## Issue Description
When using a Nintendo Switch Pro Controller via USB on a MATE Desktop environment, the controller functions correctly before suspending the system. However, upon waking the computer from suspend mode, the controller fails to work in applications like SuperTux or RetroArch. This issue does not occur with DualSense controllers.

## Steps to Reproduce
1. Plug the Nintendo Switch Pro Controller into the computer via USB.
2. Open and navigate in an application such as SuperTux or RetroArch to confirm the controller is working.
3. Close the application.
4. Suspend the desktop environment by navigating to **Menu -> Shut Down... -> Suspend** in MATE.
5. Wake up the computer by pressing any key.
6. Attempt to navigate in SuperTux or RetroArch again.

## Expected Behavior
The Nintendo Switch Pro Controller should continue to function after waking up from suspend mode.

## Actual Behavior
The controller does not respond in applications after waking from suspend mode.

## Additional Notes
- This issue is specific to the Nintendo Switch Pro Controller and does not affect DualSense controllers.
- The problem is not present in Ubuntu MATE 22.04 (Linux 5.19) and later versions, which have kernel support for this controller.
- It is recommended to re-evaluate this issue in Trisquel 12 as it may lack the necessary kernel support.

## Environment
- **Operating System:** MATE Desktop Environment
- **Controller:** Nintendo Switch Pro Controller
- **Applications Tested:** SuperTux, RetroArch
- **Kernel Version:** (Specify if known)

Please investigate and advise on potential solutions or workarounds for this issue.

davidhedlund, 2024-10-02:
Note: I could not evaluate if Bluetooth is also affected in this issue due to an [unrelated bug](https://github.com/bluez/bluez/issues/673). Please have a look.

SuperSamus, 2024-10-02:
The dkms module isn't maintained anymore, use the driver included in the kernel by default (this has been fixed in Linux 6.7, https://github.com/torvalds/linux/commit/95ea4d9fd385fe335b989f22d409df079a042b7a).

davidhedlund, 2024-10-02:
> The dkms module isn't maintained anymore

https://github.com/nicman23/dkms-hid-nintendo says "For any questions or bug reports, please refer to [hid_nintendo](https://github.com/DanielOgorchock/linux)."

That is why I submitted my bug report here. Do you think that nicman23 should change that text?

> use the driver included in the kernel by default (this has been fixed in Linux 6.7, [torvalds/linux@95ea4d9](https://github.com/torvalds/linux/commit/95ea4d9fd385fe335b989f22d409df079a042b7a)).

Some distros are using older versions of the kernel. Thank you.