DanielOgorchock/linux #56: ECHTPower - Tons of dropped IMU reports

donob4n, 2025-03-23:
Hi, 

I got a ECHTPower gamepad which seems oriented to nintendo switch and thanks to 'hid_nintendo' it works via bluetooth but I get constant messages like:

```
7250.259555] nintendo 0005:057E:2009.000E: compensating for 4 dropped IMU reports
[ 7250.259564] nintendo 0005:057E:2009.000E: delta=1564 avg_delta=241
[ 7255.869521] nintendo 0005:057E:2009.000E: compensating for 9 dropped IMU reports
[ 7255.869530] nintendo 0005:057E:2009.000E: delta=2749 avg_delta=241
[ 7260.555557] nintendo 0005:057E:2009.000E: compensating for 17 dropped IMU reports
[ 7260.555566] nintendo 0005:057E:2009.000E: delta=4686 avg_delta=241
[ 7279.179696] nintendo 0005:057E:2009.000E: compensating for 75 dropped IMU reports
[ 7279.179704] nintendo 0005:057E:2009.000E: delta=18624 avg_delta=241
[ 7342.860551] nintendo 0005:057E:2009.000E: compensating for 262 dropped IMU reports
[ 7342.860560] nintendo 0005:057E:2009.000E: delta=63681 avg_delta=241
[ 7348.828572] nintendo 0005:057E:2009.000E: compensating for 21 dropped IMU reports
[ 7348.828582] nintendo 0005:057E:2009.000E: delta=5635 avg_delta=241
[ 7351.043548] nintendo 0005:057E:2009.000E: compensating for 5 dropped IMU reports
[ 7351.043556] nintendo 0005:057E:2009.000E: delta=1664 avg_delta=241
[ 7360.845581] nintendo 0005:057E:2009.000E: compensating for 6 dropped IMU reports
[ 7360.845590] nintendo 0005:057E:2009.000E: delta=1997 avg_delta=241
[ 7364.547551] nintendo 0005:057E:2009.000E: compensating for 5 dropped IMU reports
[ 7364.547559] nintendo 0005:057E:2009.000E: delta=1684 avg_delta=241
[ 7370.563572] nintendo 0005:057E:2009.000E: compensating for 23 dropped IMU reports
[ 7370.563582] nintendo 0005:057E:2009.000E: delta=6016 avg_delta=241
[ 7373.428598] nintendo 0005:057E:2009.000E: compensating for 5 dropped IMU reports
[ 7373.428607] nintendo 0005:057E:2009.000E: delta=1679 avg_delta=241
[ 7376.395544] nintendo 0005:057E:2009.000E: compensating for 4 dropped IMU reports
[ 7376.395553] nintendo 0005:057E:2009.000E: delta=1480 avg_delta=241
```

donob4n, 2025-03-23:
It seems a duplicated of https://github.com/DanielOgorchock/linux/issues/33

OctopusRage, 2026-03-12:
anyway if u run into echtpower not connected as pro controller you can try this bash script to fix it: https://gist.github.com/OctopusRage/b5bc1a62512fcc4ddc4f7e7d6c4367dc